Koenkk/zigbee2mqtt #12844: Tuya water leak does not reports periodically its status

galbip, 2022-06-16:
### What happened?

Tuya TS0207_water_leak_detector does reports its status only when the "leak sensor" changes status. 
No periodical reports if "leak sensor" status is not changing.

### What did you expect to happen?

A water leak should periodically, (at least once a day) send its status regardless there is a water leak or not. 
This assure the sensor is working and ready to send alarm when water leak will happen.

### Zigbee2MQTT version

 1.25.1 commit: 3f6a137

### Adapter firmware version

0x26720700

### Adapter

Conbee II

AlexKay88, 2022-06-22:
What MQTT message do you expect?
water_leak value stays the same until device reports a change. If device does not report anything, Z2M can't do anything.

galbip, 2022-06-22:
any MQTT message that would confirm the device is functional and alive regardless water_leak status.

From Tuya documentation: 
[https://developer.tuya.com/en/docs/iot-device-dev/tuya-zigbee-water-sensor-access-standard?id=K9ik6zvon7orn] 
it is reported the device send every 4hrs the battery level, but I do not see this in the log.

AlexKay88, 2022-06-22:
Your refer to docs how the device should be implemented in theory, but this device does obviously not completely follow the developer guidelines.
If you do not see expected communication in debug log, then the device does simply not send it.
State attribute `last_seen` is updated with timestamp of last received Zigbee message from device.
For Z2M device availability checks, see https://www.zigbee2mqtt.io/guide/configuration/device-availability.html

galbip, 2022-06-23:
correct. As long as the sensor has the water_leak status = true, it sends every 5 minute a message.
The issue is when water_leak status = false and I'm not getting any messages that tell me the sensor is alive.

As per your previous suggestion, I have now enabled the device availability check on Z2M setting 8hrs timeout, this should trigger an "offline" MQTT message for the sensor if no zigbee messages are received in last 8hrs (regardless water_leak status)

closed 2022-07-31 by github-actions[bot]