Koenkk/zigbee2mqtt #12847: z2m crashes from time to time

okastl, 2022-06-16:
### What happened?

z2m crashes sometimes. Normal Log reveals nothing interesting. I have redirected stderr to a file to get a hint what happens.
It isn't much, but I hope it helps:

```
Error: Read 0x00178801098363af/11 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 11836 - 11 - 82 - 6 - 1 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

```

Some timeout really shouldn't cause the whole process to terminate.

### What did you expect to happen?

Not crashing would be great. ;)

### How to reproduce it (minimal and precise)

Let it run for a long time and hope... I hope the stderr output gives some hint how to trigger the problem?

### Zigbee2MQTT version

1.25.2-dev commit: 820c20af

### Adapter firmware version

20220507

### Adapter

Slaesh CC2652RB stick

uspass, 2022-07-01:
Check if usb is not turned off to save power

okastl, 2022-07-01:
> Check if usb is not set to be turned off to save power

On a Raspberry Pi 3B? How?

okastl, 2022-07-01:
After two weeks of flawless operation, z2m crashed again:

```
Error: Read 0x001788010b53a259/11 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> ZDO - extRouteDisc - {"dstAddr":33327,"options":0,"radius":30}' failed with status '(0xc7: NWK_TABLE_FULL)' (expected '(0x00: SUCCESS)'))
    at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

```

NWK_TABLE_FULL. Hmm.

Koenkk, 2022-07-01:
Can you provide some more logging of this? (not just the error)

okastl, 2022-07-01:
> Can you provide some more logging of this? (not just the error)

This is just the output of stderr when the node.exe process terminated. Unfortunately I didn't have zigbee-herdsman logging enabled, as the log would grow very big. z2m was running 14 days without problem, I am afraid the log would be huge.
Can you give me a hint to provoke the error sooner?

Koenkk, 2022-07-02:
@okastl I don't need the herdsman debug logging, just the zigbee2mqtt debug logging is enough

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

okastl, 2022-07-02:
> I don't need the herdsman debug logging, just the zigbee2mqtt debug logging is enough

Attached is the debug log until z2m crashed. I hope this helps. I don't see anything out of the ordinary.

[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/9033469/log.zip)

okastl, 2022-07-02:
This is the problem. The crash happened Yesterday 2022-07.01 at 22:07:02, as this is the date of the stderr output file. The last log entry is dated 2022-07-01 22:06:59
It looks like the error didn't make it into the log before z2m crashed.

King-anderson, 2022-07-03:
@Koenkk 
I have same issue. version 1.25.2, master branch. after some time it crashed and doesn't work.
from line 4073.
when I want to make switch on/off, it doesn't respond.

[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/9034873/logs.txt)

Koenkk, 2022-07-04:
@King-anderson in your case the adapter crashes or disconnects. What system are you running z2m on and what adapter/firmware do you have?

King-anderson, 2022-07-07:
@Koenkk 
No, I said about the Zigbee2MQTT which it worded before and doesn't have the crashing problem, not about communicating the serial port instantaneously with Zigbee2MQTT. 
but about the serial port communication: yes. two applications don't been allowed to open the same serial port. but, as a practice. when Zigbee2MQTT is running fine when I open the port in a python code, the port is opened and I can send and receive data.  I may prevent Zigbee2MQTT to communicate using serial but after I close the port in Python code and see the Zigbee2MQTT log, it's working fine. It seems that zigbee2mqtt opens and closes the serial port when wants to communicate.

Koenkk, 2022-07-07:
@King-anderson but if I understood https://github.com/Koenkk/zigbee2mqtt/issues/12847#issuecomment-1176117588 correctly, the problem goes away when you stop running your Python program?

Koenkk, 2022-07-07:
@King-anderson since you are running the default CC2530 fw, can you try switching to the source routing one? This fw improves stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

Koenkk, 2022-07-08:
@King-anderson if reflashing is needed to get the adapter working again, it means it got corrupted. This is a common problem with the CC2530 and happens due to the outdated hardware and stack (not maintained by TI anymore). I suggest upgrading to a more powerful adapter: https://www.zigbee2mqtt.io/guide/adapters/#recommended

King-anderson, 2022-07-09:
> @King-anderson since you are running the default CC2530 fw, can you try switching to the source routing one? This fw improves stability: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

If I reflash the adapter with source routing firmware, should I rejoin the devices?

Koenkk, 2022-07-09:
@King-anderson no

King-anderson, 2022-07-12:
@Koenkk OK, but I didn't have any problem with previous versions of z2m. I will test and feedback to you. thank you.

King-anderson, 2022-07-17:
@Koenkk 
UPDATE:

I have done a few tests with various versions of z2m (1-8-2, 1-25-2, and 1-26-0) on the same hardware (CC2530 serial adapter) and about 40 ZigBee devices including 30 routers. in all tests, after some time and sending a command to ZNP, It freezes and doesn't respond to commands and doesn't send received data from ZigBee devices (receive timeout error and sometimes table-nwk-route). the ZNP needs to be programmed to operate finely. 

is there any way to estimate the maximum number of devices to prevent the ZNP crash?

Koenkk, 2022-07-17:
> is there any way to estimate the maximum number of routers/devices to prevent the ZNP crash?

Depends a lot on the network form, amount of routers, packets transmitted etc. But with the zigstar you should be fine for (100+)

King-anderson, 2022-07-24:
UPDATE: 

> 
I had done a lot of extensive tests, and the final result: I think it doesn't relate to memory and it seems to adapter firmware bug. I tested with fewer devices and the crash problem occurred after 2 days. with z2m v1-8-2 and v1-25-2. with just 8 routers and 4 end-devices. there is an ambiguity about the reason for crashing the adapter which should be reprogrammed to work fine again.

King-anderson, 2022-07-24:
>  I have done some tests with various versions of z2m (1-8-2, 1-25-2, and 1-26-0) on the same hardware (CC2530 serial adapter) 

UPDATE:
 I had done a lot of extensive tests, and the final result: I think it doesn't relate to memory and it seems to adapter firmware bug. I tested with fewer devices and the crash problem occurred after 2 days. with z2m v1-8-2 and v1-25-2. with just 8 routers and 4 end-devices. there is an ambiguity about the reason for crashing the adapter which should be reprogrammed to work fine again.

@Koenkk 
can you get me a few roadmaps to check the adapter z-stack source code to find the problem?

Koenkk, 2022-07-24:
@King-anderson you can find instructions here on how to compile and retrieve the firmware code: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/COMPILE.md

Looks like a bug in the zstack code to me but chances that this will be fixed by TI is low since this stack is EOL.

closed 2022-09-03 by github-actions[bot]