Koenkk/zigbee2mqtt #12849: [New device support]: Beok TGR-85-ZB Heating Thermostate

itsolon, 2022-06-16:
### Link

https://de.aliexpress.com/item/1005003329761199.html?spm=a2g0o.detail.1000014.12.3e6cdafbUAsahz&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40000.267768.0&scm_id=1007.40000.267768.0&scm-url=1007.40000.267768.0&pvid=d376da29-0f6f-43bb-b2c3-ba2b51e5c147&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40000.267768.0,pvid:d376da29-0f6f-43bb-b2c3-ba2b51e5c147,tpp_buckets:668%232846%238108%23133&pdp_ext_f=%7B%22sku_id%22%3A%2212000028683267243%22%2C%22sceneId%22%3A%2230050%22%7D&pdp_npi=2%40dis%21EUR%21%2121.35%21%21%21%21%21%40211b5e2b16554116246071855eab17%2112000028683267243%21rec

### Database entry

{"id":9,"type":"Router","ieeeAddr":"0xa4c138e9f8e3d5ce","nwkAddr":5844,"manufId":4417,"manufName":"_TZE200_2ekuz3dz","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":" i>*\u0013.m>*e/m>*\u0012","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1655411874831,"defaultSendRequestWhen":"immediate"}

### Comments

This Device is already supported in ZHA but due to so many devices which are supported at zigbee2mqtt i switched over to zigbee2mqtt solving all my problems with more devices supported .. but one device is left.. this Thermostate
its summer.. but im planning to use this device in winter for my heating.
hopefully we can get it to work ? what do you think?

itsolon, 2022-07-07:
no one who can help?

vladyspavlov, 2022-07-26:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/3840

closed 2022-09-03 by github-actions[bot]