Koenkk/zigbee2mqtt #12853: [New device support]:

itbod365, 2022-06-17:
### Link

https://www.philips-hue.com/en-gb/p/hue-white-ambiance-buckram-triple-spotlight/8719514339125

### Database entry

{"id":45,"type":"Router","ieeeAddr":"0x0017880108cf7795","nwkAddr":62860,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"5047331P6","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,64515,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"5047331P6","manufacturerName":"Philips","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20211210","swBuildId":"1.93.7"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":3,"dateCode":"20211210","swBuildId":"1.93.7","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1655463977984,"defaultSendRequestWhen":"immediate"}

### Comments

Unsupported - Philips Hue Double Buckram - 5047331P6

Home Assistant Core 2022.6.6
Zigbee2MQTT Current version: 1.25.2-1

Have tried dev branch but didn't appear to work, you have a number of supported "Philips Buckram" bulbs already just not "5047331P6"

Many Thanks

### Supported color modes

Buckram Hue White Ambiance - 350 Lumen 50/60Hz - 2200k - 6500k

### Color temperature range

2022-06-17 12:19:31Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}

Koenkk, 2022-06-17:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-06-17 by Koenkk

itbod365, 2022-06-17:
Thanks Koenkk much appreciated in getting both 5047231P6 & 5047331P6 added!