Koenkk/zigbee2mqtt #12855: Interview failed of all devices, CC2531 stick

drummeroo, 2022-06-17:
### What happened?

After new installation of domoticz and Zigbee I have problem with my devices (now added only 2 of 5) - all of them are in "unsupported" state and their status is "interview failed". All devices worked properly before.

[image]

### What did you expect to happen?

Fix that

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20201127

### Adapter

 usb-Texas_Instruments_TI_CC2531_USB_CDC

### Debug log

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/8927347/debug.txt)

drummeroo, 2022-06-21:
Hello,

maybe someone could be able to help me with this ?

drummeroo, 2022-06-21:
I tried everything from this webpage

drummeroo, 2022-06-23:
But it worked earlier for each devices (sonoff, tuya, xiaomi etc.). I have this problem after update.

Koenkk, 2022-06-24:
it seems the device does not stay connected for some reason, did I understand correctly that you started a new network? If yes, make sure to disconnect all Zigbee devices from power. Then connect one device to power and pair it, etc.

joaoachando, 2022-06-24:
Having the same problem that @Raizelian, while trying to connect my IKEA Tradfri lights (also tried some smart light switches) it fails the interview with the log: 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8977604/log.txt)
It appears that the adapter gets disconnected for some reason.
After that, the zigbee2mqtt restarts and the device appears as unsupported.

Koenkk, 2022-06-26:
@joaoachando the log you provided is very short and doesn't contain the herdsman debug logging (so I cannot determine the cause in your case). 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

joaoachando, 2022-06-26:
Hey @Koenkk thanks for the reply!
Here is the log with the herdsman debug turned on:
[debug_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8987068/debug_log.txt)
Its the full log, from starting to trying to add a device and then automatically restarting.

drummeroo, 2022-06-27:
> it seems the device does not stay connected for some reason, did I understand correctly that you started a new network? If yes, make sure to disconnect all Zigbee devices from power. Then connect one device to power and pair it, etc.

I probably did it (removed the battery) but ok, I will try it tomorrow and let you know.

Koenkk, 2022-06-27:
@joaoachando I don't see any interview attempt in this log (seems it did not pair at all?)

joaoachando, 2022-06-29:
Hey, @Koenkk today I updated to the dev branch, still fails but I think it starts (or at least tries to) the interview process, so maybe the logs will help.
[debuglod2.txt](https://github.com/Koenkk/zigbee2mqtt/files/9014098/debuglod2.txt)

Koenkk, 2022-07-01:
@joaoachando I think your CC2531 somehow got corrupted (common for the CC2531 and CC2530), reflashing the latest firmware should fix it.

drummeroo, 2022-07-04:
And what with me ?
We've strayed a bit from my topic :)

Koenkk, 2022-07-04:
@drummeroo I think you've tried all the standard things now; pairing close to the coordinator; reflashing coordinator firmware; power down all other zigbee devices before pairing. In that case I need a sniff when trying to pair the device: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

drummeroo, 2022-07-12:
So I need to flash my USB stick ?

Koenkk, 2022-07-12:
yes

drummeroo, 2022-07-21:
Ok, reflash CC2531 helped.

closed 2022-07-21 by Koenkk