Koenkk/zigbee2mqtt #12860: After reading Aquara e1 roller blinds (Xiaomi ZNJLBL01LM) i get error

ThrawnJL, 2022-06-18:
### What happened?

After my aquara e1 was frozen and not working i removed it from z2m and reset it and added it again to z2m. All works now BUT I keep getting following error in z2m each 23 seconds or so.

Error 2022-06-18 08:16:55Publish 'set' 'state' to 'Rullegardin Køkken' failed: 'Error: Write 0x54ef441000329afb/1 genAnalogOutput({"85":{"value":0,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 6266 - 1 - 210 - 13 - 4 after 10000ms)'

### What did you expect to happen?

no errors and e1 just working

### How to reproduce it (minimal and precise)

wait 23 seconds and error will occur.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20211217

### Adapter

sonoff 3.0 usb plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8933070/log.txt)

Otnow, 2022-06-18:
Try restarting _z2m_.

ThrawnJL, 2022-06-18:
Ahhh I’ll try that 
I actually thought I did that when restarting homeassistant directly but apparently not 😳 I am sorry and thank you for your help !!

norbertbxl, 2022-06-18:
Try this :
remove device aqara 
remove coordinator end wait 5 minut
put it back coordinator 
restart z2m
add devices aqara

closed 2022-06-18 by ThrawnJL

Otnow, 2022-06-18:
@ThrawnJL, ok 👍

If you don't mind, then to improve support, please write the [following debug log](https://github.com/Koenkk/zigbee2mqtt/issues/12647#issuecomment-1143623963).

ThrawnJL, 2022-06-18:
> @ThrawnJL, ok 👍
> 
> If you don't mind, then to improve support, please write the [following debug log](https://github.com/Koenkk/zigbee2mqtt/issues/12647#issuecomment-1143623963).

It worked after a restart (Thanks again) 
Is that debug thingie-my-jig necessary?

Otnow, 2022-06-18:
> Is that debug thingie-my-jig necessary?

It is not required for your problem, but this debug log will help improve support for _ZNJLBL01LM_, so if you have the ability, please write it and post it here.

ThrawnJL, 2022-06-18:
> > Is that debug thingie-my-jig necessary?
> 
> It is not required for your problem, but this debug log will help improve support for _ZNJLBL01LM_, so if you have the ability, please write it and post it here.

Ok I will give it a try when I can find some time again :)