Koenkk/zigbee2mqtt #12863: Crash when trying to add new device

NicolaiPetri, 2022-06-18:
### What happened?

Trying to add new Nordtronic box dimmer, zigbee2mqtt crashes with the following error:
Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":19,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,20,6,0,187,97,0,73,11,2,226,240,1,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')
    at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
    
   
I don't know (yet) if it's because I also have 2 wiser powertags that doesn't seem to pair correctly and it keeps spamming network when pairing is enabled.

### What did you expect to happen?

Pairing should succeed.

### How to reproduce it (minimal and precise)

Start pairing when nordtronic is in pairing mode. Doesn't matter which device(s) i select for pairing.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20211217

### Adapter

zig-a-zig-aah

### Debug log

rent phase a","state_topic":"zigbee2mqtt/0x00000000e2020b49","unique_id":"0x00000000e2020b49_current_phase_a_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current_phase_a }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b49/current_phase_b/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b49/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b49"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b49"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","name":"0x00000000e2020b49 current phase b","state_class":"measurement","state_topic":"zigbee2mqtt/0x00000000e2020b49","unique_id":"0x00000000e2020b49_current_phase_b_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current_phase_b }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b49/current_phase_c/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b49/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b49"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b49"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","name":"0x00000000e2020b49 current phase c","state_class":"measurement","state_topic":"zigbee2mqtt/0x00000000e2020b49","unique_id":"0x00000000e2020b49_current_phase_c_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current_phase_c }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b49/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b49/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b49"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b49"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x00000000e2020b49 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00000000e2020b49","unique_id":"0x00000000e2020b49_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b49/last_seen/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b49/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b49"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b49"},"device_class":"timestamp","enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:clock","name":"0x00000000e2020b49 last seen","state_topic":"zigbee2mqtt/0x00000000e2020b49","unique_id":"0x00000000e2020b49_last_seen_zigbee2mqtt","value_template":"{{ value_json.last_seen }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b79/power/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b79/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b79"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b79"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","name":"0x00000000e2020b79 power","state_class":"measurement","state_topic":"zigbee2mqtt/0x00000000e2020b79","unique_id":"0x00000000e2020b79_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b79/power_apparent/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b79/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b79"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b79"},"enabled_by_default":true,"name":"0x00000000e2020b79 power apparent","state_topic":"zigbee2mqtt/0x00000000e2020b79","unique_id":"0x00000000e2020b79_power_apparent_zigbee2mqtt","unit_of_measurement":"VA","value_template":"{{ value_json.power_apparent }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b79/power_phase_a/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b79/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b79"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b79"},"enabled_by_default":true,"name":"0x00000000e2020b79 power phase a","state_topic":"zigbee2mqtt/0x00000000e2020b79","unique_id":"0x00000000e2020b79_power_phase_a_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power_phase_a }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b79/power_phase_b/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b79/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b79"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b79"},"enabled_by_default":true,"name":"0x00000000e2020b79 power phase b","state_topic":"zigbee2mqtt/0x00000000e2020b79","unique_id":"0x00000000e2020b79_power_phase_b_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power_phase_b }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b79/power_phase_c/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b79/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b79"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b79"},"enabled_by_default":true,"name":"0x00000000e2020b79 power phase c","state_topic":"zigbee2mqtt/0x00000000e2020b79","unique_id":"0x00000000e2020b79_power_phase_c_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power_phase_c }}"}'
Zigbee2MQTT:debug 2022-06-18 11:16:56: Received MQTT message on 'homeassistant/sensor/0x00000000e2020b79/power_factor/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00000000e2020b79/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0x00000000e2020b79"],"manufacturer":"Schneider Electric","model":"PowerTag power sensor (A9MEM1570)","name":"0x00000000e2020b79"},"device_class":"power_factor","enabled_by_default":false,"entity_category":"diagnostic","name":"0x00000 [... cut ...]
Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":19,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,20,6,0,187,97,0,73,11,2,226,240,1,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')
    at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

NicolaiPetri, 2022-06-18:
Update, it seems like it's powertags making fun. At least this happens starting pairing mode even without nordtronic dimmer connected.

Koenkk, 2022-06-18:
Can you try with the latest fw https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin ?

NicolaiPetri, 2022-07-04:
> Can you try with the latest fw https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin ?

I have updated to latest firmware and now also updated to latest Z2M, but still same issue. Crash when I enable join mode.

NicolaiPetri, 2022-07-04:
Latest log lines with updated firmware and Z2M:
`  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":40,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,37,6,0,71,202,0,73,11,2,226,240,1,0]}} +8ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,34,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,40,0,30,14,0,25,37,6,0,71,202,0,73,11,2,226,240,1,0,249] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +10ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":11738,"destendpoint":242,"srcendpoint":242,"clusterid":33,"transid":41,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[25,38,1,168,1,0,73,11,2,226,132,11,254]}} +10ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,1,218,45,242,242,33,0,41,0,30,13,25,38,1,168,1,0,73,11,2,226,132,11,254,154] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100] +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +20ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":42,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,39,6,0,218,45,0,73,11,2,226,240,1,0]}} +21ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,34,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,42,0,30,14,0,25,39,6,0,218,45,0,73,11,2,226,240,1,0,131] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,41,28] +2ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,41,28] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,41] - 28 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":41} +52ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:controller:log Green power device '{"sourceID":3791784777,"deviceID":254,"networkAddress":2889}' joined +2s
Zigbee2MQTT:debug 2022-07-04 19:03:09: Received Zigbee message from '0x00000000e2020b49', type 'commandNotification', cluster 'greenPower', data '{"commandFrame":{"applicationInfo":3,"deviceID":254,"extendedOptions":131,"gpdClientClusters":{"data":[],"type":"Buffer"},"gpdCommandIdList":{"data":[],"type":"Buffer"},"gpdServerClusters":{"data":[],"type":"Buffer"},"keyMic":0,"manufacturerID":4190,"modelID":17245,"numClientClusters":0,"numGdpCommands":0,"numServerClusters":0,"options":231,"outgoingCounter":5084185,"securityKey":{"data":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"type":"Buffer"}},"commandID":224,"frameCounter":67,"gppGddLink":255,"gppNwkAddr":11738,"options":20496,"payloadSize":12,"srcID":3791784777}' from endpoint 242 with groupID 2889
Zigbee2MQTT:info  2022-07-04 19:03:09: MQTT publish: topic 'zigbee2mqtt/0x00000000e2020b49', payload '{"ac_frequency":null,"current_phase_a":null,"current_phase_b":null,"current_phase_c":null,"device":{"friendlyName":"0x00000000e2020b49","ieeeAddr":"0x00000000e2020b49","manufacturerID":null,"model":"A9MEM1570","networkAddress":2889,"type":"GreenPower"},"energy":null,"energy_phase_a":null,"energy_phase_b":null,"energy_phase_c":null,"last_seen":"2022-07-04T17:03:09.011Z","linkquality":21,"power":null,"power_apparent":null,"power_factor":null,"power_phase_a":null,"power_phase_b":null,"power_phase_c":null,"voltage_phase_a":null,"voltage_phase_ab":null,"voltage_phase_b":null,"voltage_phase_bc":null,"voltage_phase_c":null,"voltage_phase_ca":null}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,17,118] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,17,118] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [17] - 118 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":17} +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62677,"destendpoint":242,"srcendpoint":242,"clusterid":33,"transid":43,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[25,40,1,168,1,0,73,11,2,226,132,11,254]}} +12ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,1,213,244,242,242,33,0,43,0,30,13,25,40,1,168,1,0,73,11,2,226,132,11,254,64] +13ms
Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":42,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,39,6,0,218,45,0,73,11,2,226,240,1,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')
    at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)`

Koenkk, 2022-07-06:
Could you check if this is fixed now?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

NicolaiPetri, 2022-07-08:
> Could you check if this is fixed now?
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Hi, 
I tried latest dev, and it did improve pairing and I could add new devices, but I still get same error when in pairing mode, it just seems to crash less or maybe just in a later step:
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,33,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,15,0,30,13,0,25,17,1,168,1,0,73,11,2,226,132,11,254,73] +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +21ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +22ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":23480,"destendpoint":242,"srcendpoint":242,"clusterid":33,"transid":16,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[25,16,1,168,1,0,73,11,2,226,132,11,254]}} +22ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,1,184,91,242,242,33,0,16,0,30,13,25,16,1,168,1,0,73,11,2,226,132,11,254,129] +21ms
  zigbee-herdsman:controller:log Green power device '{"sourceID":3791784777,"deviceID":254,"networkAddress":2889}' joined +2s
  zigbee-herdsman:controller:log Green power device '{"sourceID":3791784777,"deviceID":254,"networkAddress":2889}' joined +29ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,225,242,15,219] +906ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,225,242,15,219] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [225,242,15] - 219 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":225,"endpoint":242,"transid":15} +948ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,13,56,254,3,68,128,0,242,14,59] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,13,56,254,3,68,128,0,242,14,59] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,13] - 56 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":13} +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,14,59] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,14] - 59 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":14} +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100,254,3,68,128,0,242,16,37] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100,254,3,68,128,0,242,16,37] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0} +933ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,16,37] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,16] - 37 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":16} +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - extNwkInfo - {} +934ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,37,80,117] +935ms
  zigbee-herdsman:controller:log Green power device '{"sourceID":3791784777,"deviceID":254,"networkAddress":2889}' joined +735ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,101,80,0,0,9,255,231,0,0,204,239,45,136,188,147,17,83,0,0,0,0,0,0,0,0,11,220] +7ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,101,80,0,0,9,255,231,0,0,204,239,45,136,188,147,17,83,0,0,0,0,0,0,0,0,11,220] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 3 - 5 - 80 - [0,0,9,255,231,0,0,204,239,45,136,188,147,17,83,0,0,0,0,0,0,0,0,11] - 220 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extNwkInfo - {"shortaddr":0,"devstate":9,"panid":59391,"parentaddr":0,"extendedpanid":"0x531193bc882defcc","parentextaddr":"0x0000000000000000","channel":11} +9ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":17,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,18,6,0,208,137,0,73,11,2,226,240,1,0]}} +9ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,34,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,17,0,30,14,0,25,18,6,0,208,137,0,73,11,2,226,240,1,0,35] +8ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,17,118] +453ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,17,118] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [17] - 118 +0ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":17} +455ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
[... cut ...]
Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":17,"options":0,"radius":30,"len":14,"data":{"type":"Buffer","data":[25,18,6,0,208,137,0,73,11,2,226,240,1,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')
    at Object.func (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

Koenkk, 2022-07-09:
Does z2m really crash in terms of stops or is it just the error that is being logged?

NicolaiPetri, 2022-07-09:
Hi. 
i was sure it stopped but have not been able to reproduce that. So I might have seen it on an old terminal or similar. 
still my PowerTags doesn’t work, but I’ve probably not had any crashes after switching to latest dev. 
i don’t know how to troubleshoot power tag issue, it seems that final step in pairing fails and the power tag specific handler that parses payloads seems to never be called. I tried to add some console.log commands there since I thought it was a parsing issue of package.

NicolaiPetri, 2022-07-09:
But this issue can probably be closed.

Koenkk, 2022-07-10:
The `BUFFER_FULL` is still an error I want to investigate, this error seems to be on the firmware side. I expect that TI soon releases a new firmware so I will ping you once it is there.

Koenkk, 2022-07-24:
@NicolaiPetri can you try the 20220724 fw?

NicolaiPetri, 2022-07-25:
Hi @Koenkk ,
I just tried but seems to give same issues, and pairing my wiser powertags still doesn't complete:
Zigbee2MQTT version
1.26.0-dev commit: [316daee4](https://github.com/Koenkk/zigbee2mqtt/commit/316daee4)
Coordinator type
zStack3x0
Coordinator revision
20220724

0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')'
2022-07-25T08:50:26.556Z zigbee-herdsman:controller:greenpower onZclGreenPowerData failed with error 'Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":64,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[25,11,1,168,1,0,73,11,2,226,132,11,254]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')'
2

Koenkk, 2022-07-25:
Can you try with:
[znp_CC26X2R1_LAUNCHXL_tirtos_ticlang.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/9183350/znp_CC26X2R1_LAUNCHXL_tirtos_ticlang.hex.zip)

NicolaiPetri, 2022-07-26:
> Can you try with: [znp_CC26X2R1_LAUNCHXL_tirtos_ticlang.hex.zip](https://github.com/Koenkk/zigbee2mqtt/files/9183350/znp_CC26X2R1_LAUNCHXL_tirtos_ticlang.hex.zip)

Yes, but seems like identical issue:
2022-07-26T16:07:01.686Z zigbee-herdsman:controller:greenpower onZclGreenPowerData failed with error 'Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":75,"options":0,"radius":30,"len":13,"data":{"type":"Buffer","data":[25,76,1,168,1,0,73,11,2,226,132,11,254]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')'

I'm at the point where I'm probably gonna try to buy a spare coordinator and see if it's related to specific devices or settings on my network. It will also make tests a bit easier so I don't fear destroying my setup at home :)

LuukML, 2022-07-27:
hi @Koenkk and @NicolaiPetri ill start following this thread, we are encountering the same error in about 18 dongles/gateways. (my hobby exploded a bit..). with us it started to happen when we swapped the raspberry Pi + TS dev pcb for the ethernet adapter. 
our setup is:
- TubesZB dongel (https://www.tubeszb.com/product/cc2652_poe_coordinator/47)
- docker latest  
- networks max 50 devices

the error we get is : 

Publish 'set' 'brightness' to '13/office/15/group/light_01' failed: 'Error: Command 4 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":50}) failed (SREQ '--> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000004","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":8,"transid":161,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[17,75,4,254,50,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))'

The network will respond slow and after a reboot of the dongle and docker it will be back to normal and respond fast via mqtt but the UI will still have a hard time and be slow. @SamAlhabash can also have a look if we need extra info. Our workaround now will be rebooting the networks every night to keep them from crashing.

closed 2022-09-04 by github-actions[bot]