Koenkk/zigbee2mqtt #12867: Device Remove Not working after Failed to remove device Errohttps://github.com/Koenkk/zigbee2mqtt/labels/problemr

msadeghz, 2022-06-18:
### What happened?

Hi
I received below log from zigbee2mqtt when I want to remove a battery powered device:
`Zigbee2MQTT:error 2022-06-18 12:19:55: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device '0xxxxxxxxxxxxxxxxx' (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'
Zigbee2MQTT:debug 2022-06-18 12:19:56: Error: Failed to remove device '0xxxxxxxxxxxxxxxxx' (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)
    at Bridge.removeEntity (/home/root/zigbee2mqtt/lib/extension/bridge.ts:557:15)
    at Bridge.onMQTTMessage (/home/root/zigbee2mqtt/lib/extension/bridge.ts:122:14)
Zigbee2MQTT:info  2022-06-18 12:19:56: MQTT publish: topic 'zigbee2mqtt/bridge/request/device/remove', payload '{"data":{},"error":"Failed to remove device '0xxxxxxxxxxxxxxxxxxxxx' (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)","status":"error"}'
Zigbee2MQTT:info  2022-06-18 12:19:56: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{},"error":"Failed to remove device '0xxxxxxxxxxxxxxxxx' (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)","status":"error"}'`

but after this error, when I send remove command I don't see its log in zigbee2mqtt log. it seems that the topic 
`zigbee2mqtt/bridge/request/device/remove` topic is unsubscribed!

### What did you expect to happen?

I what to see remove command log on zigbee2mqttt log output.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211210

### Adapter

CC2530

benjR, 2022-06-18:
@msadeghz Not an expert, but I think you have to force remove the device now

msadeghz, 2022-06-18:
when you publish a command, you should see a log says a command received.
but after seeing this failure log, I don't see any log about receiving remove command.

Koenkk, 2022-06-18:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

msadeghz, 2022-06-20:
Hi @Koenkk 
you can see debug log in below image:

[image]

after I send "remove command" and it fails, when I send "remove command" again, I didn't see any log indicating an MQTT command received. it seems that the remove command topic is unsubscribed.

Koenkk, 2022-06-20:
I see the issue, but I cannot explain it yet. can you provide your `configuration.yaml`?

msadeghz, 2022-06-21:
> I see the issue, but I cannot explain it yet. can you provide your `configuration.yaml`?

thank you. it's relevant that I get `zigbee-herdsman` error connection with the zigbee adapter (my zigbee adapter is serial CC2530). I have to reprogram CC2530  or reset it to resolve the issue. 

[image]

King-anderson, 2022-06-26:
@Koenkk 
did you check my logs? is there any bug in zigbee2mqtt?

Koenkk, 2022-06-26:
Can you make sure this also happens with z2m 1.25.2?

King-anderson, 2022-07-03:
the issue was in 1.25.1. it is resolved in 1.25.2.
thank you

closed 2022-07-03 by King-anderson

karlhuther, 2024-02-01:
I am running 1.35.1 and getting the same error but with with a SONOFF motion sensor. This is happening through the frontend of Z2M. What was changed in the configuration.yaml to make this go away? I am also using a CC2652P for my coordinator.