Koenkk/zigbee2mqtt #12870: Siterwell GS361A-H04 Thermostatic Radiator Valve not show valve postion

Andy-Hennah, 2022-06-18:
### What happened?

I successfully paired a Siterwell GS361A-H04 TRV with z2m.  Values are coming through to the web UI for all exposed attributes except **position** and **running_state**, these are displaying **N/A**.
This issue was noted on #8813 after it was closed.  It is still happening.
[image]

### What did you expect to happen?

I was expecting the value of **position** to display the position of the valve as the valve changes, from **0**% (closed) to **100**% (fully open).
I was expecting **running_state** to change from **idle** to **heat** depending on whether the valve is open or not.
Note, the valve_detection value is set to "ON".

### How to reproduce it (minimal and precise)

Make changes to the heating set point and hear the TRV's valve opening and closing.  

### Zigbee2MQTT version

1.25.2 commit: a252914e

### Adapter firmware version

20210120

### Adapter

Electrolama zzh

### Debug log

[logX.txt](https://github.com/Koenkk/zigbee2mqtt/files/8934036/logX.txt)

lbida, 2022-06-19:
Have the same issue.

Andy-Hennah, 2022-07-24:
Hi,
I was wondering if there's anyone out there who can see the valve position property working?

Andy-Hennah, 2022-08-24:
This is still a problem.

Andy-Hennah, 2022-08-24:
@Ibida and @TheLexus are you still seeing this as an issue?

Prouser123, 2022-08-30:
Hi, I've just installed these and am having the same issue. My battery percentage is also reporting as `N/A %`, but I suspect that is because the percentage has not changed and so has not been published/updated.

james-fry, 2022-09-09:
I have looked at the siterwell manual (PFA) and also the github for zigpy/ZHA support for this TRA and as far as I can see, it does not support value position. Its an error in the Z2M device js to expose valve position.

What the TRV does support is valve fault detection (boolean) - see page 15 in the manual.

You can see the ZHA supported functions here:
https://user-images.githubusercontent.com/49618193/143578078-1c3d43e2-56b6-40d1-8dac-801096880e3e.PNG

(from [here](https://github.com/zigpy/zha-device-handlers/issues/1180))

[Siterwell-ZigBee-Radiator-Thermostat-GS361A-H04-User-Manual.pdf](https://github.com/Koenkk/zigbee2mqtt/files/9533516/Siterwell-ZigBee-Radiator-Thermostat-GS361A-H04-User-Manual.pdf)

Andy-Hennah, 2022-09-11:
Thanks @james-fry 
So, is this documentation wrong then?
https://www.zigbee2mqtt.io/devices/GS361A-H04.html

peteh, 2022-09-24:
Seems like it. 
It does not have valve position and also the running_state is N/A on my device: 
[image]

eljoest, 2022-09-24:
Same for me, no `position` and no ```
`running_state`. Just bought mine as "Nedis Thermostat ZBHTR10WT".

peterkappelt, 2022-09-26:
I've bought these thermostats as well and ran in the same issue. Apparently, my model (Nedis ZBHTR10WT) doesn't even expose it's valve position, unfortunately. I've paired it with a Tuya gateway and checked it's available DPs (names taken from Tuya IoT Dashboard):
```
- 17 (State of the window)
- 2 (Set temperature)
- 18 (Window check)
- 3 (Current temperature)
- 19 (State of the valve) -> I'm not yet sure what this DP indicates
- 4 (Mode) -> manual, auto, freeze
- 20 (Valve testing)
- 21 (Battery capacity)
- 102 (TempSet) -> don't know what that's for, no messages traced so far
- 7 (Child Lock)
- 27 (Temperature correction)
```
There doesn't seem to be a "valve position" DP at all 😢

Prouser123, 2022-09-30:
Hi, from packet capture of my TCP Smart variant of the Siterwell GS361A-H04 I can see the following 'unknown' datapoints:
```
1 (binary?)
4 (seen values ranging from 0-3)
13 (hex? values of 0 and 16)
14 (binary?)
```
Thanks,

James

cc: @james-fry

peteh, 2022-10-31:
*unstale*

Still in issue: 
[image]

State message: 
[image]

jekkos, 2022-11-08:
I own this valve as well, currently seeing this in my logs
`tuyaThermostat: Unrecognized DP #17 with data {"dp":17,"datatype":4,"data":{"type":"Buffer","data":[0]}}`

Alph4d0g, 2022-12-02:
+1 got the same problem here.
Unfortunately I currently got no time to dig into this, but for what it's worth I'm at least preventing a stale topic =)

jekkos, 2022-12-17:
does the temperature reading work well for everyone here? The reading on mine always get stuck to some value after a week. I then need to reset the network key and repair them to make it work again. Beside this the link quality reading do seem to be updated, so it's only temperature that is stuck.

fstap, 2023-01-15:
> does the temperature reading work well for everyone here? The reading on mine always get stuck to some value after a week. I then need to reset the network key and repair them to make it work again. Beside this the link quality reading do seem to be updated, so it's only temperature that is stuck.

Same problem for me, this has been going on for almost two months now I think. Pretty annoying, though I had no time to investigate the issue

closed 2023-02-22 by github-actions[bot]

GamerClassN7, 2023-10-25:
Hi, Any news ?

Jezza34000, 2023-11-24:
+1 same problem for me, following this thread :-)

jekkos, 2023-11-24:
In the end I was able to make one trv log its temperature consistently. Pairing these devices remains a headache. I always end up having to reset my network key.

Also the plastic mount easily breaks. I found a 3d part that can be glued to it as a replacement.

tml89, 2023-12-19:
+1 same problem for me

yggster, 2024-01-04:
+1 same problem. Following thread :)

tbrasser, 2024-01-12:
+1 

also I guess no way to use any external temperature without resorting to better-thermostat?

meydominic, 2024-09-24:
+1 any news on this?

```
{
    "away_mode": "OFF",
    "battery": 100,
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 26.5,
    "linkquality": 255,
    "local_temperature": 21.5,
    "preset": "manual",
    "system_mode": "heat",
    "valve_detection": "ON",
    "window_detection": "ON",
    "position": null,
    "running_state": null
}
```

Soulfly999, 2025-05-06:
any news on this? The problem still exists?