Koenkk/zigbee2mqtt #12879: Paulmann Licht GmbH - CCT light is not supported any more

fintmu, 2022-06-19:
### What happened?

I updated Zigbee2MQTT from version 1.18.1-1 to 1.25.2-1 and one of my lights went to "Not supported" status. This light was working just fine with old version. Why support for this light has been removed?

Paulmann Licht GmbH - CCT light

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

20220219

### Adapter

Electrolama zzh

Koenkk, 2022-06-20:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/12701

closed 2022-06-20 by Koenkk

fintmu, 2022-06-20:
I don't see how this is duplicate with other brand device? I tried same ext_converter for this light with no luck. It didn't work even I changed manufacturerID and name.

The light doesn't expose anything and this is from the log:
warning 2022-06-20 22:18:49Received message from unsupported device with Zigbee model 'CCT light ' and manufacturer name 'Paulmann Licht GmbH '
warning 2022-06-20 22:18:49Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Here's my device info from database:
{"id":64,"type":"Router","ieeeAddr":"0x00158d0001dd7258","nwkAddr":41622,"manufId":4509,"manufName":"Paulmann Licht GmbH ","powerSource":"Mains (single phase)","modelId":"CCT light ","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,4,3,6,8,5,768,4096],"outClusterList":[25],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":200,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20160301","swBuildId":"1218-0909","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1655751069641}

Koenkk, 2022-06-21:
I see, this is because of the trailing space: `CCT light `. Added it now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)