Koenkk/zigbee2mqtt #12881: Upgraded from CC2531 to CC2652p (sonoff zigbee 3.0 Dongle Plus)

VanQz, 2022-06-20:
### What happened?

Zigbee2MQTT fails to start, i have read the website on how to fix, I use by /dev/serial/by-id however it keeps failing to start.
My cc2531 i believe have died and since i got the adapter awhile ago wanting to upgrade have have yet to do so, its what im using to change it now.

What I have done:
Flashed the latest firmware for cc2652p with https://www.youtube.com/watch?v=4jqQCxjlRDU
Changed the configuration port to path /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_.xxxxxxxxxxxx-port0
Ran the following command
1. test -w /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_xxxxxxxxxxxxxxxx-if00-port0 && echo success || echo failure

2. test -w /dev/ttyUSB0&& echo success || echo failure

First command passed but second command failed. Earlier in the day Second command passed unsure why.
Restarted the machine.

After all that was done it still shows the following error code:
[image]

### Zigbee2MQTT version

1.25.1

### Adapter firmware version

20220125

### Adapter

SONOFF ZIGBEE 3.0 USB DONGLE PLUS

Koenkk, 2022-06-20:
What is the file name of the firmware you flashed? This is the correct one: https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20220219.zip

xrs444, 2022-06-27:
Did the same with the above firmware and it worked until I rebooted, then the same problem. Running in Docker on a RasPi 3B. Probably same as https://github.com/Koenkk/zigbee2mqtt/issues/12887

closed 2022-08-04 by github-actions[bot]