Koenkk/zigbee2mqtt #12883: [New device support]: TS0601 PIR sensor with illuminance

SEVENID, 2022-06-20:
### Link

https://aliexpress.com/item/1005003891641858.html

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0xa4c138b42ca376c4","nwkAddr":18311,"manufId":4417,"manufName":"_TZE200_f1pvdgoh","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":51,"65508":0,"appVersion":66,"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1655718768631,"defaultSendRequestWhen":"immediate"}

### Comments

Seems like #12364. 
Using current docker image. 

Log is full of (maybe illumination level) - reports about every 11s
```
debug 2022-06-20 09:58:48Received Zigbee message from '0xa4c138b42ca376c4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,232],"type":"Buffer"},"datatype":2,"dp":101}],"seq":13057}' from endpoint 1 with groupID 0
debug 2022-06-20 09:58:59Received Zigbee message from '0xa4c138b42ca376c4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,109],"type":"Buffer"},"datatype":2,"dp":101}],"seq":13313}' from endpoint 1 with groupID 0
debug 2022-06-20 09:59:10Received Zigbee message from '0xa4c138b42ca376c4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,189],"type":"Buffer"},"datatype":2,"dp":101}],"seq":13569}' from endpoint 1 with groupID 0
debug 2022-06-20 09:59:32Received Zigbee message from '0xa4c138b42ca376c4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,204],"type":"Buffer"},"datatype":2,"dp":101}],"seq":14081}' from endpoint 1 with groupID 0
```
If i move:
```
debug 2022-06-20 10:13:47Received Zigbee message from '0xa4c138b42ca376c4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":24833}' from endpoint 1 with groupID 0
debug 2022-06-20 10:13:48Received Zigbee message from '0xa4c138b42ca376c4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,3],"type":"Buffer"},"datatype":2,"dp":101}],"seq":25089}' from endpoint 1 with groupID 0
debug 2022-06-20 10:25:56Received Zigbee message from '0xa4c138b42ca376c4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1}],"seq":31233}' from endpoint 1 with groupID 0
```

IvanUpa, 2022-06-22:
PIR State:
{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "1",
    "startTime": 1655749811612,
    "endTime": 1655922611613,
    "projectCode": "p1655920058432xk9rcy",
    "sourceId": "eu1651682053718E4RJe",
    "sourceType": "4",
    "deviceId": "bfd2306d18752dae89ri31",
    "pageStartRow": "",
    "region": "EU"
}

Battery level:
{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "4",
    "startTime": 1655749909012,
    "endTime": 1655922709013,
    "projectCode": "p1655920058432xk9rcy",
    "sourceId": "eu1651682053718E4RJe",
    "sourceType": "4",
    "deviceId": "bfd2306d18752dae89ri31",
    "pageStartRow": "",
    "region": "EU"
}

Illuminance Value:
{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "101",
    "startTime": 1655749956941,
    "endTime": 1655922756941,
    "projectCode": "p1655920058432xk9rcy",
    "sourceId": "eu1651682053718E4RJe",
    "sourceType": "4",
    "deviceId": "bfd2306d18752dae89ri31",
    "pageStartRow": "",
    "region": "EU"
}

Charge state:
{
    "startRowId": "",
    "pageNo": 1,
    "pageSize": 10,
    "code": "102",
    "startTime": 1655749994793,
    "endTime": 1655922794793,
    "projectCode": "p1655920058432xk9rcy",
    "sourceId": "eu1651682053718E4RJe",
    "sourceType": "4",
    "deviceId": "bfd2306d18752dae89ri31",
    "pageStartRow": "",
    "region": "EU"
}

SEVENID, 2022-06-26:
Meanwhile, got the battery report:

```
Received Zigbee message from '0xa4c138b42ca376c4', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,100],"type":"Buffer"},"datatype":2,"dp":4}],"seq":3074}' from endpoint 1 with groupID 0
```
There are no ideas how to request it from the device, it itself reports very rarely.

Temporary external converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
            {
                modelID: 'TS0601',
                manufacturerName: '_TZE200_f1pvdgoh',
            },
        ],
    model: 'TS0601_Haozee_PIR',
    vendor: 'TuYa',
    description: 'Haozee PIR sensor with illumination',
    fromZigbee: [fz.ignore_basic_report,
                //fz.tuya_data_point_dump,
                {
                    cluster: 'manuSpecificTuya',
                    type: ['commandDataReport'],
                    convert: (model, msg, publish, options, meta) => {
                            const result = {};
                            for (const dpValue of msg.data.dpValues) {
                                    const dp = dpValue.dp;
                                    const value = tuya.getDataValue(dpValue);
                                    switch (dp) {
                                    case 1:
                                            result.occupancy = (value === 0);
                                            break;
                                    case 4:
                                            result.battery = value;
                                            break;
                                    case 101:
                                            result.illuminance = value;
                                            result.illuminance_lux = value;
                                            break;
                                    default:
                                            throw new Error(`Unhandled datapoint: ${JSON.stringify(dpValue)}`);
                                    }
                            }
                        return result;
                    }
                }
        ],
    toZigbee: [
        tz.tuya_data_point_test,
        ],
    exposes: [
        e.occupancy(), e.illuminance(), e.illuminance_lux(), e.battery()
    ],
};

module.exports = definition;
```

The status of the motion sensor is updated once every 2 minutes.
The sensor settings are unknown, if any. 
Illumination and lux seem to be identical, at least for tuya, but the raw data changes drastically (from 400 to 1000) with no real change in illuminance (sensor under a lamp with no other light sources). 
Datapoint 102 is not in use - no messages have been received containing it, nor any status or change in external power. Perhaps a rechargeable battery is required? The device was delivered to me with a non-rechargeable battery.
The real model code is still unknown, the manufacturer is only listed on the sale page.

IvanUpa, 2022-07-03:
Updating to the latest firmware and with a converter from SEVENID the sensor sends "null" for all attributes
`"linkquality":69,"occupancy":null`

heggink, 2022-07-26:
The above seems to work fine for me with the exception that the device seems to be capped at 1000 lux giving me the impression that there must be some kind of max setting (like some of these tuya tem/himidity devices have)...

closed 2022-09-05 by github-actions[bot]

SEVENID, 2023-06-04:
Updating z2m to 1.31.0 breaks my external converter with such errors:

```
        if (converter.options) {
                      ^
/app/node_modules/zigbee-herdsman-converters/index.js:79
Using '/app/data' as data directory
    at start (/app/index.js:106:18)
    at new Controller (/app/lib/controller.ts:84:58)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:79:23)
TypeError: Cannot read properties of undefined (reading 'options')
```
I don't know what exactly broke and, unfortunately, I don't have the opportunity to find out yet.

Bacchus777, 2023-09-18:
Here is my working with Z2M 1.33 converter

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_f1pvdgoh',
        },
    ],
    model: 'TS0601_Haozee_PIR',
    vendor: 'TuYa',
    description: 'Haozee PIR sensor with illumination',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
    exposes: [e.occupancy(), e.illuminance(), e.battery()
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'occupancy', tuya.valueConverter.trueFalse0], 
            [101, 'illuminance', tuya.valueConverter.raw],
            [4, 'battery', tuya.valueConverter.raw],
        ],
    },
};

module.exports = definition;

```

upais, 2023-09-19:
> Here is my working with Z2M 1.33 converter

It works, thanks!

Koenkk, 2023-09-19:
thanks @Bacchus777, added the device for out-of-the-box support!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

whcrg, 2023-12-01:
Ihave seemingly the exact same device but reports different manufacturer
Device '0xa4c13878e686f0cb' with Zigbee model 'TS0601' and manufacturer name '_TZE200_s6hzw8g2' is NOT supported...

Seems to be working at least mostly with the converter Bacchus777 had above with just manyfacturer edited.

Unsure if the brightness works correct.

whcrg, 2023-12-01:
Brightnes/illuminance works but only updates when movement (as above people also had with the "manufName":"_TZE200_f1pvdgoh"). So this device confirmed to work with the same coverter.