Koenkk/zigbee2mqtt #12887: Zigbee2MQTT not starting with “Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)”

enzodesena, 2022-06-20:
### What happened?

Hi everyone. This has been discussed elsewhere, but it looks like most discussions appear to be stale, so here goes. 

I am running HA on Raspberry Pi 3, with a Sonoff Universal Zigbee 3.0 USB Dongle Plus. The dongle works well with ZHA. However, I need Zigbee2MQTT. I follow the instructions but get the dreaded  "Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)". The full log is here: https://pastebin.com/gEwq17K2

I have tried the following, without success:
- Added the following lines to the mqtt section: "user: mqtt-user and password: my-password"
- Connected the dongle through a USB-extension cable
- Updated the Sonoff Dongle with the latest firmware
- Installed the Zigbee2MQTT Edge (i.e. developer branch) instead of Zigbee2MQTT
- Used as port /dev/serial/by-id/.... instead of /dev/ttyUSB0
- Tried two different power supplies. 
- Followed the instructions on ttps://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html , including: (a) checking that the port is correct; (b) Verifying that the user running Zigbee2MQTT as has write access to the port (I am assuming here that the user would be root. 
- All the usual stuff: restarting HA, restarting raspberry pi, reinstalling add-on. 

My full configuration is here: https://pastebin.com/52mTVPmg

Anything else I could try? Thanks in advance.

### How to reproduce it (minimal and precise)

Here is the configuration file:

```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: '-d -d'
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
  user: mqtt-user
  password: my-password
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_3e2fe9a18412ec119a4b20c7bd930c07-if00-port0
```

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

Latest

### Adapter

Sonoff Universal Zigbee 3.0 USB Dongle Plus

### Debug log

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/socat.sh
[21:31:51] INFO: Socat not enabled, marking service as down
cont-init: info: /etc/cont-init.d/socat.sh exited 0
cont-init: info: running /etc/cont-init.d/zigbee2mqtt.sh
cont-init: info: /etc/cont-init.d/zigbee2mqtt.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun socat (no readiness notification)
services-up: info: copying legacy longrun zigbee2mqtt (no readiness notification)
[21:32:05] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.25.2 start
> node index.js
Zigbee2MQTT:info  2022-06-20 21:32:32: Logging to console and directory: '/config/zigbee2mqtt/log/2022-06-20.21-32-31' filename: log.txt
Zigbee2MQTT:info  2022-06-20 21:32:32: Starting Zigbee2MQTT version 1.25.2 (commit #unknown)
Zigbee2MQTT:info  2022-06-20 21:32:32: Starting zigbee-herdsman (0.14.34)
Zigbee2MQTT:error 2022-06-20 21:32:53: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-06-20 21:32:54: Failed to start zigbee
Zigbee2MQTT:error 2022-06-20 21:32:54: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-06-20 21:32:54: Exiting...
Zigbee2MQTT:error 2022-06-20 21:32:55: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:102:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
[21:33:06] INFO: Handing over control to Zigbee2mqtt Core ...

tomazzive, 2022-06-20:
That is exactly the same experience. I have been troubleshooting for hours and days. 
Exact same config, exact same error. Sonoff stick is correctly flashed with TI.

enzodesena, 2022-06-20:
@tomazzive Same here. I just posted this on HA community, and it turns out it was the simplest thing: uninstall ZHA..

tomazzive, 2022-06-20:
I already uninstalled ZHA. restarted HA. reinstalled the zigbee2mqqtt addons en broker. etc
It doesn't work. I reflashed my stick just now. But again; nothing.

King-anderson, 2022-06-21:
I have this issue too. it even not resolved if I reflash the zigbee adapter (with serial CC2530 adapter).

hocomadvies, 2022-06-22:
I have the same issue have been looking for a solution for days now! even bought a new stick hoping that would help. topic #12821    I'm using a cc2652 Stick and  the latest FW and Zigbee2mqtt 1.2.25

xrs444, 2022-06-26:
Same here, worked on first boot after replacing C2531, however no Tradfri bulbs would join, but other devices worked fine. Rebooted to see if that would clear it, and now I get the same as above. Reflashed, rebooted, checked the port, and replaced the extension cable. 

Zigbee2MQTT is running in the latest docker container as root pn a RPi 3B. I replaced the original controller as the network was getting larger and I could not reconnect a Tradfri bulb, so I suspect I actually have two issues now...

Cossey, 2022-06-27:
I've been noticing the adapter has been crashing too - and yes it is a Sonoff Dongle! It was stable for months, but only recently in the past 3-4 weeks it has been crashing. Took a while to find the culprit. However I checked for the USB autosuspend (it was enabled on the x64 system, debian) and disabled it (weird as it was not causing an issue before). Resolved the issue for a few weeks, then just tonight the dongle crashed on me again - super annoying!

It's hard to tell what it is because I updated both the firmware and the z2m at the same time. However, z2m, even with the regular updates I do, hasn't really caused an issue like this before.

I have firmware `20220219 zStack3x0` running on `1.25.1`. I have a suspicion there could be a bug somewhere in the latest firmware affecting our specific devices.

I might downgrade to previous firmware version and give that a go. @Koenkk is there anything specific we can try to get a better diagnosis?

Koenkk, 2022-06-27:
To debug this problem further I need to have the herdsman debug logging starting from the point where you start z2m until it fails.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

tomazzive, 2022-06-27:
Hi Koenkk, thank you for your reply. 
I got it working. I spent 12 hours in solving this issue.
Solution: flash the sonoff dongle with the coordinator instead with the router hex file. 
FML!

xrs444, 2022-06-29:
@Koenkk  I'll try and get the logs for you, but it may not be until the weekend. If I grab the container console I should see the STDOUT output right?

Koenkk, 2022-07-01:
@xrs444 yes, you can use `docker logs CONTAINER_NAME > container.log 2>&1`

King-anderson, 2022-07-05:
zigbee-herdsman#556

King-anderson, 2022-07-05:
Koenkk/zigbee-herdsman#556

King-anderson, 2022-07-05:
maybe the same issue with corresponding logs. 
Koenkk/zigbee-herdsman#556

closed 2022-08-12 by github-actions[bot]

djansen1987, 2022-10-12:
Same issue, Sometimes it resolves by re-seating the dongle and restart de z2m addon in HA. Any new on how to solve this issue ?

hydazz, 2022-10-22:
Same issue here with the Egony Stick V4. Everything works perfectly then randomly the stick becomes unresponsive. Needs to be re-seated to get it to work again

```
Zigbee2MQTT:info  2022-10-22 12:17:27: Logging to console and directory: '/config/log/2022-10-22.12-17-27' filename: log.txt
Zigbee2MQTT:info  2022-10-22 12:17:27: Starting Zigbee2MQTT version 1.28.0 (commit #unknown)
Zigbee2MQTT:info  2022-10-22 12:17:27: Starting zigbee-herdsman (0.14.62)
Zigbee2MQTT:error 2022-10-22 12:17:47: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-10-22 12:17:47: Failed to start zigbee
Zigbee2MQTT:error 2022-10-22 12:17:47: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-10-22 12:17:47: Exiting...
Zigbee2MQTT:error 2022-10-22 12:17:47: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
```

coen1111, 2022-11-14:
I have exactly the same issue here.

bachoo786, 2022-11-28:
same issue here today using the latest image of z2m on docker compose. Can someone please help?

vivobg, 2022-12-21:
I had the same issue, after updating my Hass OS, Hass itself and z2m, and then rebooting the system. It turned out ZHA somehow got turned on again, and required a manual disable. After that, z2m was able to start successfully.
[image]

Triskelen, 2023-01-08:
i gave the same error 

`[17:30:16] INFO: Preparing to start...
[17:30:16] INFO: Socat not enabled
[17:30:17] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2023-01-08 17:30:19: Logging to console and directory: '/config/zigbee2mqtt/log/2023-01-08.17-30-19' filename: log.txt
Zigbee2MQTT:info  2023-01-08 17:30:19: Starting Zigbee2MQTT version 1.29.1 (commit #unknown)
Zigbee2MQTT:info  2023-01-08 17:30:19: Starting zigbee-herdsman (0.14.83)
Zigbee2MQTT:error 2023-01-08 17:30:38: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-01-08 17:30:38: Failed to start zigbee
Zigbee2MQTT:error 2023-01-08 17:30:38: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-01-08 17:30:38: Exiting...
Zigbee2MQTT:error 2023-01-08 17:30:38: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:107:5)`

I don't have the zha integration installed never have only used z2m when the device first boots it's fine. 
after x amount of time it's gone. When i use MQTT explorer it connect's just fine and the adapter is stil up and running
and still posting.

Aidam, 2023-03-12:
> adapter: deconz

Ahhh thanks!

  This did help after hours of troubleshooting

Aidam, 2023-03-12:
> Ever tried to set the serial `adapter: deconz`
> 
> Worked for me with ConBeeII

Aidam, 2023-03-12:
> Ever tried to set the serial `adapter: deconz`
> 
> Worked for me with ConBeeII

 Thanks! This finally helped after hours of troubleshooting.

lukefor, 2023-03-27:
Same issue here, after around a couple of days my CC2531 falls over until replugged with the following error:

```
$ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.30.2 start
> node index.js

Zigbee2MQTT:debug 2023-03-27 18:01:14: Loaded state from file ******/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2023-03-27 18:01:14: Logging to console and directory: '******/zigbee2mqtt/data/log/2023-03-27.18-01-14' filename: log.txt
Zigbee2MQTT:debug 2023-03-27 18:01:14: Removing old log directory '******/zigbee2mqtt/data/log/2023-03-27.17-44-14'
Zigbee2MQTT:info  2023-03-27 18:01:14: Starting Zigbee2MQTT version 1.30.2 (commit #cdf62ea5)
Zigbee2MQTT:info  2023-03-27 18:01:14: Starting zigbee-herdsman (0.14.96)
Zigbee2MQTT:debug 2023-03-27 18:01:14: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"******/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"******/zigbee2mqtt/data/database.db.backup","databasePath":"******/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[******],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"******/zigbee2mqtt/data/database.db","databaseBackupPath":"******/zigbee2mqtt/data/database.db.backup","backupPath":"******/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +20s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +0ms
  zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload +256ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +253ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +1s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352 +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +16s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +16s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2023-03-27 18:02:10: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-03-27 18:02:10: Failed to start zigbee
Zigbee2MQTT:error 2023-03-27 18:02:10: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-03-27 18:02:10: Exiting...
Zigbee2MQTT:error 2023-03-27 18:02:10: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (******/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (******/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (******/zigbee2mqtt/lib/zigbee.ts:58:27)
    at Controller.start (******/zigbee2mqtt/lib/controller.ts:101:27)
    at start (******/zigbee2mqtt/index.js:107:5)
```

It was working reliably until I updated zigbee2mqtt (was a year or so out of date)

pbassut, 2024-01-13:
If you're using Dongle-E, upgrade the firmware [here](https://darkxst.github.io/silabs-firmware-builder/)