Koenkk/zigbee2mqtt #12888: Sonoff zigbee 3.0 USB dongle doesn't stay connected as a ROUTER

ashokaiyar, 2022-06-20:
### What happened?

Hello, I have been using zigbee2mqtt for more than 2 years. I have a total of 84 devices with 19 routers. This is my present setup:

zigbee2mqtt version 1.25.2 
Coordinator: Slae.sh CC2652RB zigbee stick running coordinator firmware dated 20220219 (CC2652RB_coordinator_20220219.hex)

I needed a new router for my front yard, decided to flash a spare "Sonoff Zigbee 3.0 USB Dongle Plus" as a router using the latest router firmware available for that dongle, i.e "CC1352P2_CC2652P_launchpad_router_20220125.hex".

This stick pairs perfectly and shows up as a powered router in the web interface (port 8080). However, after pairing, it stops communicating with the coordinator. And after a while gets marked as being "Offline".

I am flashing the router firmware using the cc2538-bsl.py Python script. I have tried the flag "--bootloader-sonoff-usb", and also tried flashing by manually pressing the boot button. In both cases, there are no errors during flashing, and the outcome is the same - pairs easily, but will not stay connected.

I would greatly appreciate some help in solving this and using it as a router.

Many thanks!

### What did you expect to happen?

The Sonoff Zigbee 3.0 USB stick would pair stably as a router.

### How to reproduce it (minimal and precise)

Flash router firmware (CC1352P2_CC2652P_launchpad_router_20220125.hex) to Sonoff Zigbee 3.0 USB stick.
Put zigbee2mqtt into pairing mode. The stick will pair very rapidly.
Wait 20 minutes. The stick will be marked as offline (basically it stops communicating after initial pairing).

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB stick

trongcongbk, 2022-06-21:
Hello,
I am using a lot of "Sonoff Zigbee 3.0 USB Dongle Plus" as a router (more than 30 devices). I flash same firmware as you are using. 

And i see all of "Sonoff Zigbee 3.0 USB Dongle Plus" working pertfectly. 
To keep the device comunicate with coordinator, I configure the device to report periodically to the coordinator (around 4 minutes), you can try this way!
[image]

[image]

ashokaiyar, 2022-06-21:
@trongcongbk - thank you very much! I had forgotten to change that! Much appreciated!!