Koenkk/zigbee2mqtt #12918: Zigbee2MQTT container crashes

BDV77, 2022-06-23:
### What happened?

Zigbee2MQTT container crashes

### What did you expect to happen?

Container not to crash...

### How to reproduce it (minimal and precise)

This happens once every couple of days, not really deterministic as far as I could tell. 

### Zigbee2MQTT version

1.25.2 commit: 3c5854f

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Debug log

Zigbee2MQTT:info  2022-06-21 21:04:58: MQTT publish: topic 'zigbee2mqtt/LIGHT_GU10_02', payload '{"brightness":125,"color_mode":"color_temp","color_temp":370,"linkquality":255,"power_on_behavior":null,"state":"OFF","update":{"state":"available"},"update_available":true}'
Zigbee2MQTT:info  2022-06-21 21:05:18: MQTT publish: topic 'zigbee2mqtt/LIGHT_E27_03', payload '{"brightness":247,"color":{"h":29,"hue":29,"s":89,"saturation":89,"x":0.4918,"y":0.415},"color_mode":"color_temp","color_temp":427,"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-06-21 21:05:18: MQTT publish: topic 'zigbee2mqtt/LIGHT_E27_01', payload '{"brightness":247,"color":{"h":29,"hue":29,"s":89,"saturation":89,"x":0.4918,"y":0.415},"color_mode":"color_temp","color_temp":427,"linkquality":196,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Error: Read 0x0017880100f8b169/11 genOnOff(["onOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received)
    at DeconzAdapter.sendZclFrameToEndpoint (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:655:23)
    at Endpoint.sendRequest (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:299:20)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:472:28)
    at Object.<anonymous> (/app/lib/extension/bind.ts:472:29)

BDV77, 2022-06-23:
I also noticed that building of the map takes very long, more than 10 minutes and it is not loaded yet

Koenkk, 2022-06-24:
What kind of device is the one with ieeAddr `0x0017880100f8b169`?

BDV77, 2022-06-24:
Thanks for the quick response!

BDV77, 2022-06-26:
Hi Koenkk,

Is this enough information or can I provide more? Container seems to crash on about daily basis...

Koenkk, 2022-06-26:
Should be fixed now, can you confirm?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

BDV77, 2022-06-27:
Just upgraded to 1.25.2-dev commit: [e291687], I will let you know if this is stable for me.

Thanks!

BDV77, 2022-07-01:
Hi,

The setup has been working stable for a week now, so from my perspective it can be considered solved.

Thanks a lot for the solution and quick response!!

Koenkk, 2022-07-01:
Cool 😄

closed 2022-07-01 by Koenkk