Koenkk/zigbee2mqtt #12924: Aqara Water Leak Sensor (SJCGQ11LM)

Jens-Wymeersch, 2022-06-24:
### What happened?

In short, I don't get a value for water_leak, battery_low and power_outage_count. 

### What did you expect to happen?

Nothing special happened, just don't get this reading

### How to reproduce it (minimal and precise)

I have 2 devices like this, both have the same problem.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

ZZH!

ashokaiyar, 2022-06-25:
I have several of these with the same version of zigbee2mqtt (1.25.2) and my coordinator (slaesh's stick) is flashed with the same coordinator version.

The sensors pair and report fine. Here's what the state looks like:

```
{
    "battery": 100,
    "battery_low": false,
    "device_temperature": 26,
    "last_seen": 1656118444933,
    "linkquality": 83,
    "tamper": false,
    "temperature": 26,
    "voltage": 3025,
    "water_leak": false
}
```

Jens-Wymeersch, 2022-06-25:
I re-paired the sensor and now all got sorted ! Thanks !

closed 2022-06-25 by Jens-Wymeersch