Koenkk/zigbee2mqtt #12925: Sfter the update wrong identification device PIR Neo Tuya PST-ZMIR01

wirszews, 2022-06-24:
### What happened?

Hi after zigbee2mqtt to 1.25.2 from 1.25.1 update recognizes PIR Neo Tuya PST-ZMIR01 as alarm NAS-AB02B2.

Removing and adding a device again has the same effect. I had to go back to the previous version 1.25.1 3f6a1378

my database before and after update
[https://files.fm/u/7945abdk5](url)

### How to reproduce it (minimal and precise)

It is enough to go back to the previous version for it to detect PIR Neo Tuya PST-ZMIR01

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20211217

### Adapter

zStack3x0

### Debug log

info  2022-06-24 08:57:49: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-06-24.08-57-49' filename: log.txt
info  2022-06-24 08:57:49: Starting Zigbee2MQTT version 1.25.2 (commit #9b2a9210)
info  2022-06-24 08:57:49: Starting zigbee-herdsman (0.14.34)
info  2022-06-24 08:57:49: zigbee-herdsman started (resumed)
info  2022-06-24 08:57:49: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"}'
info  2022-06-24 08:57:49: Currently 8 devices are joined:
info  2022-06-24 08:57:49: Window Sensor - 6 (0x00158d0007c5b875): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-06-24 08:57:49: Pokoj_Temp (0x00158d0007df5e6d): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-06-24 08:57:49: Window Sensor - 3 (0x00158d0007c57160): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
info  2022-06-24 08:57:49: Lazienka_Temp (0x00158d0007df9733): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2022-06-24 08:57:49: Lazienka_Fun (0x00124b0024c1e4ec): ZBMINI - SONOFF Zigbee two way smart switch (Router)
info  2022-06-24 08:57:49: Lazienka_Ruch (0x00124b0024e77dd1): SNZB-03 - SONOFF Motion sensor (EndDevice)
info  2022-06-24 08:57:49: Kontakt TV (0xa4c1385c2f3d2fb5): TS011F_plug_1 - TuYa Smart plug (with power monitoring) (Router)
info  2022-06-24 08:57:49: Skrzynka (0x00158d000806f523): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
warn  2022-06-24 08:57:49: `permit_join` set to  `true` in configuration.yaml.
warn  2022-06-24 08:57:49: Allowing new devices to join.
warn  2022-06-24 08:57:49: Set `permit_join` to `false` once you joined all devices.
info  2022-06-24 08:57:49: Zigbee: allowing new devices to join.
info  2022-06-24 08:57:50: Connecting to MQTT server at mqtt://localhost:1883
info  2022-06-24 08:57:50: Connected to MQTT server
info  2022-06-24 08:57:50: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
info  2022-06-24 08:57:50: Started frontend on port 0.0.0.0:8081
info  2022-06-24 08:57:50: MQTT publish: topic 'homeassistant/sensor/0x00158d0007c5b875/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007c5b875"],"manufacturer":"Xiaomi","model":"Aqara door & window contact sensor (MCCGQ11LM)","name":"Window Sensor - 6","sw_version":"3000-0001"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"Window Sensor - 6 battery","state_class":"measurement","state_topic":"zigbee2mqtt/Window Sensor - 6","unique_id":"0x00158d0007c5b875_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2022-06-24 08:57:50: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0007c5b875/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007c5b875"],"manufacturer":"Xiaomi","model":"Aqara door & window contact sensor (MCCGQ11LM)","name":"Window Sensor - 6","sw_version":"3000-0001"},"device_class":"door","name":"Window Sensor - 6 contact","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/Window Sensor - 6","unique_id":"0x00158d0007c5b875_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
info  2022-06-24 08:57:50: MQTT publish: topic 'homeassistant/sensor/0x00158d0007c5b875/device_temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007c5b875"],"manufacturer":"Xiaomi","model":"Aqara door & window contact sensor (MCCGQ11LM)","name":"Window Sensor - 6","sw_version":"3000-0001"},"device_class":"temperature","enabled_by_default":true,"entity_category":"diagnostic","name":"Window Sensor - 6 device temperature","state_class":"measurement","state_topic":"zigbee2mqtt/Window Sensor - 6","unique_id":"0x00158d0007c5b875_device_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.device_temperature }}"}'
info  2022-06-24 08:57:50: MQTT publish: topic 'homeassistant/sensor/0x00158d0007c5b875/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007c5b875"],"manufacturer":"Xiaomi","model":"Aqara door & window contact sensor (MCCGQ11LM)","name":"Window Sensor - 6","sw_version":"3000-0001"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","name":"Window Sensor - 6 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/Window Sensor - 6","unique_id":"0x00158d0007c5b875_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
info  2022-06-24 08:57:50: MQTT publish: topic 'homeassistant/sensor/0x00158d0007c5b875/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007c5b875"],"manufacturer":"Xiaomi","model":"Aqara door & window contact sensor (MCCGQ11LM)","name":"Window Sensor - 6","sw_version":"3000-0001"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Window Sensor - 6 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/Window Sensor - 6","unique_id":"0x00158d0007c5b875_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info  2022-06-24 08:57:50: MQTT publish: topic 'homeassistant/sensor/0x00158d0007df5e6d/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007df5e6d"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Pokoj_Temp","sw_version":"3000-0001"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"Pokoj_Temp_battery","state_class":"measurement","state_topic":"zigbee2mqtt/Pokoj_Temp","unique_id":"0x00158d0007df5e6d_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2022-06-24 08:57:50: MQTT publish: topic 'homeassistant/sensor/0x00158d0007df5e6d/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007df5e6d"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Pokoj_Temp","sw_version":"3000-0001"},"device_class":"temperature","enabled_by_default":true,"name":"Pokoj_Temp_temperature","state_class":"measurement","state_topic":"zigbee2mqtt/Pokoj_Temp","unique_id":"0x00158d0007df5e6d_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
info  2022-06-24 08:57:50: MQTT publish: topic 'homeassistant/sensor/0x00158d0007df5e6d/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007df5e6d"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"Pokoj_Temp","sw_version":"3000-0001"},"device_class":"humidity","enabled_by_default":true,"name":"Pokoj_Temp_humidity","state_class":"measurement","state_topic":"zigbee2mqtt/Pokoj_Temp","unique_id":"0x00158d0007df5e6d_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
[... cut ...]
info  2022-06-24 09:04:46: MQTT publish: topic 'zigbee2mqtt/Kontakt TV', payload '{"child_lock":"LOCK","current":0,"energy":0.88,"indicator_mode":"off/on","linkquality":43,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"state":"available"},"update_available":null,"voltage":244}'
info  2022-06-24 09:04:46: MQTT publish: topic 'zigbee2mqtt/Kontakt TV', payload '{"child_lock":"LOCK","current":0,"energy":0.88,"indicator_mode":"off/on","linkquality":43,"power":0,"power_outage_memory":"restore","state":"OFF","update":{"state":"available"},"update_available":null,"voltage":244}'

mabosch74, 2022-06-24:
try to replace this section on /zigbee-herdsman-converters/devices/tuya.js with the follwing:
`    {
        fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dr6sduka'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_ef5xlc9q'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_vwqnz1sn'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_2b8f6cio'},
            {modelID: 'TS0202', manufacturerName: '_TZE200_bq5c8xfe'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_qjqgmqxr'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_mmtwjmaq'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_zwvaj5wy'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_bsvqrxru'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_tv3wxhcz'},
            {modelID: 'TS0202', manufacturerName: '_TYZB01_hqbdru35'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_otvn3lne'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_tiwq83wk'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_ykwcwxmz'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_hgu1dlak'},
            {modelID: 'WHD02', manufacturerName: '_TZ3000_hktqahrq'},
            {modelID: 'TS0202', manufacturerName: '_TZ3000_kmh5qpmb'}],
        model: 'TS0202',
        vendor: 'TuYa',
        description: 'Motion sensor',
        whiteLabel: [{vendor: 'Mercator Ikuü', model: 'SMA02P'},
            {vendor: 'TuYa', model: 'TY-ZPR06'},
            {vendor: 'Tesla Smart', model: 'TS0202'}],
        fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_occupancy_alarm_1_report],
        toZigbee: [],
        exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery(), e.battery_voltage()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
        },
    },`

wirszews, 2022-06-25:
thank you, it helped

closed 2022-08-03 by github-actions[bot]