Koenkk/zigbee2mqtt #12930: Please support LEDVANCE SMART + CLA60 RGBW JP

nbc8915, 2022-06-25:
### Link

https://www.amazon.co.jp/dp/B0839LKXRB?psc=1&ref_=cm_sw_r_cp_ud_dp_BCA1DHZNW3CMZV809RM5

### Database entry

from '0xf0d1b80000166af9', type 'readResponse', cluster 'genBasic', data '{"modelId":"CLA60 RGBW JP"}' from endpoint 1 with groupID 0

### Comments

Please support LEDVANCE SMART + CLA60 RGBW JP.

### Supported color modes

light (state, brightness, color_temp, color_temp_startup, color_xy, color_hs), effect, linkquality

Koenkk, 2022-06-26:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

mattcartycodes, 2022-07-03:
I just got some of these today. I'll make a converter tomorrow and do a PR. Do rgb lights require the colour temp range?

Koenkk, 2022-07-06:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-06 by Koenkk