Koenkk/zigbee2mqtt #12935: Temperature and power values stuck after zigbee2mqtt upgrade

fapgomes, 2022-06-25:
### What happened?

After an upgrade of zigbee2mqtt software, the values for temperature and power of xiaomi devices are stuck.
Attached some examples
[image]
[image]
[image]
.

### Zigbee2MQTT version

1.25.2 commit: 9b2a9210

### Adapter firmware version

20211116

### Adapter

CC2531

ashokaiyar, 2022-06-25:
Which Xiaomi devices? Perhaps your observation is related to this change?

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4275

fapgomes, 2022-06-25:
That I notice, this: https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html#xiaomi-mccgq11lm (on this is the temperature issue)
and this: https://www.zigbee2mqtt.io/devices/SSM-U01.html#xiaomi-ssm-u01 (on this is the power issue)

@ashokaiyar On the temperature, it can be what you write :) Thanks.

On the power value, you have an idea? It can be because I change the coordinator firmware to the routing firmware?

Thanks,

ashokaiyar, 2022-06-25:
Can you describe the power issue in more detail? What is the message that you expect to see? Are you doing this in node-red? If so, which node-red integration for zigbee2mqtt are you using? Can you use MQTT Explorer (or something similar) to see what values are being sent by z2m to your MQTT daemon?

fapgomes, 2022-06-25:
Hi,

 I use MQTT Explorer to see the value, and the value is stuck on 5.13W  (and I sure that is not the correct value, when I turn on the electric pump):

[image]
 
On z2m frontend the value is the same that I see in MQTT Explorer, and the same that I've on openhab and grafana.

@ashokaiyar  Based on that, you can help?

Thanks,

ashokaiyar, 2022-06-25:
There are no recent changes that should affected power reporting. Have you considered resetting and re-pairing the device? It will pair back using its existing Friendly Name and IEEE address.

fapgomes, 2022-06-25:
Humm, I think with the re-pair it worked!! Thank you very much.

closed 2022-06-25 by fapgomes