Koenkk/zigbee2mqtt #12944: [New device support]: Third Reality Smart Button 3RSB22BZ

abishur, 2022-06-26:
### Link

https://www.3reality.com/online-store/Third-Reality-Smart-Switch-Gen3-Zigbee-Version-p381658008

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0x282c02bfffe37516","nwkAddr":23303,"ma>

### Comments

I've set up an external converter and have it reporting, but I get the feeling I've over complicated it.  The example I followed had lots of buttons, but this just has a single button.  It appears to be working though.

It did throw up some error messages about the power configuration when I first joined it to the zigbee network, but it appears to be reporting 🤷‍♂️

I also couldn't figure out how to get the picture to show up, but the link included as part of this form includes a picture

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const getKey = (object, value) => {
    for (const key in object) {
        if (object[key] == value) return key;
    }
};
const bind = async (endpoint, target, clusters) => {
    for (const cluster of clusters) {
        await endpoint.bind(cluster, target);
    }
};

const fzLocal = {
    d3RSB22BZ_clicks: {
        cluster: 'genMultistateInput',
        type: ['readResponse', 'attributeReport'],
        convert: (model, msg, publish, options, meta) => {
            const button = getKey(model.endpoint(msg.device), msg.endpoint.ID);
            const lookup = {
                0: 'hold',
                1: 'single',
                2: 'double',
                255: 'release',
            };
            const clicks = msg.data['presentValue'];
            const action = lookup[clicks] ? lookup[clicks] : `many_${clicks}`;
            return {
                action: `${button}_${action}`,
            };
        },
    },
};

const definition = {
    zigbeeModel: ['3RSB22BZ'], 
    model: '3RSB22BZ', 
    vendor: 'Third Reality', 
    description: 'Smart Button', 
    supports: 'on/off, power measurement',
    fromZigbee: [fzLocal.d3RSB22BZ_clicks, fz.battery, fz.command_on, fz.command_off],
    toZigbee: [], 
    exposes: [e.switch(),e.battery(), e.battery_voltage(), e.action([
         'hold', 'single', 'double', 'release', 'on', 'off'])], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        const payload = [{
            attribute: 'batteryPercentageRemaining',
            minimumReportInterval: 0,
            maximumReportInterval: 200,
            reportableChange: 0,
        }, {
            attribute: 'batteryVoltage',
            minimumReportInterval: 0,
            maximumReportInterval: 3600,
            reportableChange: 0,
        }];
        await endpoint.configureReporting('genPowerCfg', payload);

        device.endpoints.forEach(async (ep) => {
            await bind(ep, coordinatorEndpoint, ['genMultistateInput']);
        });
    },
    endpoint: (device) => {
        return {
            button_1: 1,
            button_2: 2,
            button_3: 3,
            button_4: 4,
            button_5: 5,
            button_6: 6,
            button_7: 7,
            button_8: 8,
            button_9: 9,
            button_10: 10,
            button_11: 11,
            button_12: 12,
            button_13: 13,
            button_14: 14,
            button_15: 15,
            button_16: 16,
            button_17: 17,
            button_18: 18,
            button_19: 19,
            button_20: 20,
        };
    },
};

module.exports = definition;
```

Koenkk, 2022-06-26:
- Does removing the `    endpoint: (device) => {` part change anything to the published `action` values?
- What are the actual published `action` values?

abishur, 2022-06-27:
When I was setting up the switch I got the follow entries in my log:
`type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":31}' from endpoint 1 with groupID 0
	type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":192}' from endpoint 1 with groupID 0
	
	type 'attributeReport', cluster 'genBasic', data '{"swBuildId":"1.00.16"}' from endpoint 1 with groupID 0
	
	type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
	type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
	
	type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
	type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":2}' from endpoint 1 with groupID 0
	
	type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":0}' from endpoint 1 with groupID 0
	type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":0}' from endpoint 1 with groupID 0

	type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":255}' from endpoint 1 with groupID 0`

Single Button press was commandOn
Double Button press (had to be done slowly or else it registered as a single button press) was commandOff

Both of these also created a genMultistateInput.  Single Button Press was presentValue: 1, Double Button press was presentValue: 2, holding it was presentingValue: 0, releasing it was presentValue:255

Considering it gives the state through genMultistateInput is the standard practice typically to ignore the commandOn and CommandOff transmissions and just use a multistate converter?

Koenkk, 2022-06-27:
Yes, ignore the command on/off and use the multistate instead.

abishur, 2022-07-01:
Here is my updated external converter file working with the smart switch and without all the needless fluff.  Would it be possible to get it and the device picture added to the next release?

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    ignore_command_off_3RSB22BZ: {
        cluster: 'genOnOff',
        type: 'commandOff',
        convert: (model, msg, publish, options, meta) => null,
    },
};

const definition = {
    zigbeeModel: ['3RSB22BZ'], // The model ID
    model: '3RSB22BZ', // Vendor model number
    vendor: 'Third Reality', // Vendor of the device
    description: 'Smart Switch Gen3', // Description of the device, copy from vendor site. (only used for documentation and s>
    fromZigbee: [fz.battery, fz.ignore_command_on, fzLocal.ignore_command_off_3RSB22BZ, fz.ignore_light_brightness_report, fz>
    toZigbee: [], // None
    exposes: [e.battery(), e.battery_voltage(), e.action()], 
};

module.exports = definition;`

abishur, 2022-07-01:
One side note, when I pair it, it reports it's power source as main voltage (single phase), but it's really just battery power

Koenkk, 2022-07-01:
Seems someone else just added this device: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4402 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-09 by github-actions[bot]

Daisame, 2023-07-08:
Has this been addressed yet? I just tried to add the same device tgo my zigbee2mqtt and get the same issue.