Koenkk/zigbee2mqtt #12946: Iluminize 511.344 not publishing on/off action messages

stefa168, 2022-06-26:
### What happened?

I recently bought an Iluminize 511.344 remote, and I paired it using the [Sunricher instructions](https://www.sunricher.com/media/resources/manual/SR-ZG2819S-RGB%20Instruction%20V1.pdf).

I am reading the realtime logs from Z2M and from Mosquitto, however I'm not seeing any `action: "on"` or `action: "off"` when pressing the on/off button on the remote.

### What did you expect to happen?

I referred to the [supported device list](https://www.zigbee2mqtt.io/devices/511.344.html#action-enum=) and it looks like an on/off action should be emitted when I press the corresponding button, however no such action is emitted over MQTT.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-06-26:
Are the other actions emitted?

stefa168, 2022-06-26:
> Are the other actions emitted?

Yes, they are, I just checked with the remote

Koenkk, 2022-06-26:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZG2819S-RGBW'],
    model: '511.344',
    vendor: 'Iluminize',
    description: 'Zigbee handheld remote RGBW 4 channels',
    fromZigbee: [fz.battery, fz.command_move_to_color, fz.command_move_to_color_temp, fz.command_move_hue,
        fz.command_step, fz.command_recall, fz.ZG2819S_command_on, fz.ZG2819S_command_off],
    exposes: [e.battery(), e.action([
        'color_move', 'color_temperature_move', 'hue_move', 'hue_stop', 'brightness_step_up', 'brightness_step_down',
        'recall_*', 'on', 'off'])],
    toZigbee: [],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {ep1: 1, ep2: 2, ep3: 3, ep4: 4};
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed
- Make sure the configure succeeds, to make sure; go to z2m frontend -> wakeup device -> press yellow reconfigure button and make sure it says "configure succeeded"

stefa168, 2022-06-26:
Ok, I followed your instructions (the configuration succeded), however there is no on/off action transmitted.

If it can be of any use, I checked with the debug log and (even without your external converter) Z2M prints

```
debug 2022-06-26 18:12:55Received Zigbee message from '0x9035eafffebcf504', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug 2022-06-26 18:12:57Received Zigbee message from '0x9035eafffebcf504', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
```

When clicking the on/off button on the remote (once for the first message, and another time for the second)

Koenkk, 2022-06-27:
This happens because the messages which don't have groupID == 46337 (https://github.com/Koenkk/zigbee-herdsman-converters/blob/a88c04ce95062e78c09a356bb94dc8a3959ec616/converters/fromZigbee.js#L5180) 

@[averbeck](https://github.com/averbeck) since you've added this, what would be the consequences of removing this?

stefa168, 2022-07-17:
@Koenkk Is there any way I can test the change myself, just to check if it can help with my issue?

Koenkk, 2022-07-17:
@stefa168 since I didn't get a reply, I pushed a fix, please check.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

stefa168, 2022-07-27:
Hi, sorry for the delay @Koenkk 
I am trying to run the dev version with the HA addon. I have set it up exactly like the release version, however I'm getting an error on startup:
```
[22:26:53] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.26.0-dev start
> node index.js
/app/node_modules/zigbee-herdsman-converters/index.js:91
        if (converter.options) {
                      ^
TypeError: Cannot read properties of undefined (reading 'options')
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:91:23)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:108:18)
```

I'm not sure what I'm doing wrong, do you have any suggestion?

Koenkk, 2022-07-28:
@stefa168 can you remove your external converter? Since it is fixed directly in z2m this is not needed anymore.

stefa168, 2022-07-29:
Thanks, I confirm it works now:

```
info 2022-07-29 14:57:06MQTT publish: topic 'zigbee2mqtt/0x9035eafffebcf504', payload '{"action":"on_ep1","battery":50,"brightness_ep1":127,"brightness_ep2":223,"elapsed":34680,"linkquality":10}'
info 2022-07-29 14:57:06MQTT publish: topic 'zigbee2mqtt/0x9035eafffebcf504', payload '{"action":"","battery":50,"brightness_ep1":127,"brightness_ep2":223,"linkquality":10}'
info 2022-07-29 14:57:06MQTT publish: topic 'zigbee2mqtt/0x9035eafffebcf504/action', payload 'on_ep1'
```

closed 2022-07-29 by stefa168