Koenkk/zigbee2mqtt #12947: LIDL/Livarno Smart LED Light Bar mixed white light

ferropolis, 2022-06-26:
### What happened?

If you choose an color (red, blue, etc.) and change afterwards the brightness from full to another brightness all white LED are also aktivated.

Look at my video at 35 seconds and 45 seconds: https://youtu.be/2IP8AcydQfg?t=35

### What did you expect to happen?

I expect if I change the brightness the color remains clear and do not mixed up whit the white LED.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20211217

### Adapter

Zonoff Zigbee 3.0 USB plus

ferropolis, 2022-08-11:
@Koenkk Did you see any chance to help me?

alterschlabberanesi, 2022-11-19:
Not stale

Koenkk, 2022-11-20:
What is the device model?

TimB87, 2022-11-20:
According to the devices product page it's the following values.

Zigbee Model: TS0505B
Model: [14149505L/14149506L](https://www.zigbee2mqtt.io/devices/14149505L_14149506L.html#lidl-14149505l%252F14149506l)

Koenkk, 2022-11-20:
Currently we use the standard Zigbee ZCL converters, but it seems the devices works a little bit different. Can anyone sniff the traffic when controlling this device via the TuYa gateway?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

TimB87, 2022-11-29:
@Koenkk or anybody: Does sniffing only work with the mentioned usb sticks? I have a CC2652P myself, but I might get my hands on a EZSP Stick (Sonoff) for this.

RunnicFusion, 2023-01-22:
Any update on this?

RunnicFusion, 2023-01-22:
Why thumb down @TimB87?

RubenKelevra, 2024-09-01:
This issue persists. @Koenkk please reopen. Clearly not stale.

RubenKelevra, 2024-09-03:
Someone got the original hub and willing to take a peek at the traffic or the firmware, regarding firmware update URLs?

RubenKelevra, 2025-01-14:
@Koenkk I don't think this one is stale

closed 2025-03-24 by github-actions[bot]