Koenkk/zigbee2mqtt #12952: [Feature request]:  Lonsonho  TS110E_2gang (_TZ3210_wdexaypg) - add power_outage_memory, switch type and min/max brightness level controls

gorbunovav, 2022-06-26:
### Is your feature request related to a problem? Please describe

https://www.zigbee2mqtt.io/devices/TS110E_2gang.html#lonsonho-ts110e_2gang

https://github.com/Koenkk/zigbee-herdsman-converters/blob/2faa0abbb2c65d7f49f4a32bf640706b0ea70e16/devices/lonsonho.js#L220

1. This device supports tuya power outage memory command, so we probably can just use the existing `tuya_switch_power_outage_memory` definition (works for me).
2. It also has ability to configure the switch type on per-endpoint base.
  Cluster: genLevelCtrl (0x0008)
  Command: Write Attributes (0x02)
  Attribute: 0xfc02
  Data Type: 8-Bit Unsigned Integer (0x20)
  Possible values: 0 - kick-back switch, 1 - Seesaw Toggle Switch, 2 - Seesaw Sync Switch
3. Tuya Smartlife plugin also allows to configure the minimal and maximal brightness level for each endpoint
  Cluster: genLevelCtrl (0x0008)
  Command: Write Attributes (0x02)
  Attribute: 0xfc03 - min, 0xfc04 - max
  Data Type: 16-Bit Unsigned Integer (0x21)
  Value: brightness value in percent with one decimal point converted into integer by multiplying by 10 (e.g. 76.5 => 765)
4. It also seems possible to read all these values.

### Describe the solution you'd like

Update the device converter.

### Describe alternatives you've considered

None

### Additional context

None

closed 2022-07-01 by Koenkk

ndbroadbent, 2022-09-23:
Hi @gorbunovav, do you know if Moes dimmer modules also support this switch type option? I have been looking for this for a while:  https://github.com/Koenkk/zigbee2mqtt/discussions/12684#discussion-4114534

I have a mix of the following dimmer modules:
* [Lonsonho TS110E_1gang](https://www.zigbee2mqtt.io/devices/TS110E_1gang.html)
* [Lonsonho TS110E_2gang](https://www.zigbee2mqtt.io/devices/TS110E_2gang.html#lonsonho-ts110e_2gang)
* [Moes MS-105B](https://www.zigbee2mqtt.io/devices/MS-105B.html#moes-ms-105b)
* [Moes MS-104Z](https://www.zigbee2mqtt.io/devices/MS-104Z.html#moes-ms-104z)
* [Moes MS-105Z](https://www.zigbee2mqtt.io/devices/MS-105Z.html#moes-ms-105z)

I believe they are all using tuya chips and firmwares, so I would love to know if the “switch type” option is a standard feature supported by all of these devices. Thanks!

cwvschaik, 2023-10-08:
Hey, Is there any update on the modules using TS0601 (such as Moes ones mentioned above) 
Would be great if this could be exposed there as well.