Koenkk/zigbee2mqtt #12953: CC2652RB stick Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby

boorac, 2022-06-26:
### What happened?

The add on fails to start with the error message:
```
 Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
```

# Steps taken to solve the issue:

1. Read the FAQ
2. Read related issues (e.g. https://github.com/Koenkk/zigbee2mqtt/issues/10858 and all linked issues)
3. Read Zigbee2MQTT fails to start checklist
4. I am using the latest firmware, flashed as per instructions: https://slae.sh/projects/cc2652/#zigbee2mqtt
5. Both znp_CC2652RB_20200715.bin and 20220219 versions have been flashed, on two distinct OS-es,  using the python tool https://github.com/JelmerT/cc2538-bsl
6. panID has been updated multiple times and does not match any existing pan ID
7. The USB stick is not connected directly to the PI4, but uses an USB cable
8. The bug also occurs in the dev branch

### Zigbee2MQTT version

 1.25.2-1

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB

Koenkk, 2022-06-27:
Can you try to:
- reflash your stick
- remove your `data/coordinator_backup.json`
- start z2m

It might be that the issue occurs due to the backup restore.

boorac, 2022-07-01:
It seems that your advice helped. What I did:

1. Reflashed my stick
2. Deleted all the contents from the folder `/config/zigbee2mqtt` in Home Assistant
3. Turned off all zigbee routers
4. Restarted zigbee2mqtt

Thank you for the suggestions.

boorac, 2022-09-10:
Hello,

unfortunately, the issue started again. Some context leading up to the issue:

* It seemingly started all by itself (no updates have been made on any addons, HA or the firmware of the stick)
* It first manifested the message:
```
Data request failed with error: 'No network route' (205)
```

which made me repeat the steps above. Now the error reappears:

```
Zigbee2MQTT:info  2022-09-10 14:18:39: Starting Zigbee2MQTT version 1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-10 14:18:39: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:error 2022-09-10 14:19:53: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-10 14:19:53: Failed to start zigbee
Zigbee2MQTT:error 2022-09-10 14:19:53: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-10 14:19:53: Exiting...
Zigbee2MQTT:error 2022-09-10 14:19:53: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:347:23)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:92:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
[14:19:55] INFO: Handing over control to Zigbee2MQTT Core ...
```

I do not know how to continue debugging the issue. Further context:

```
Every 2.0s: dmesg |  grep usb | tail -50  

[    5.479988] usb 1-1.3: cp210x converter now attached to ttyUSB0
[ 1111.157834] usb 1-1.3: USB disconnect, device number 3
[ 1769.140296] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[ 1769.245679] usb 1-1.3: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 1769.245712] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1769.245728] usb 1-1.3: Product: slae.sh cc2652rb stick - slaesh's iot stuff
[ 1769.245743] usb 1-1.3: Manufacturer: Silicon Labs
[ 1769.245756] usb 1-1.3: SerialNumber: 00_12_4B_00_22_98_85_38
[ 1769.257472] usb 1-1.3: cp210x converter now attached to ttyUSB0

```

Koenkk, 2022-09-11:
A same issue happened with other users, they managed to get it fixed by using Flash programmer 2 for flashing the firmware (https://www.zigbee2mqtt.io/guide/adapters/#notes)

boorac, 2022-09-11:
Flash programmer 2 does not work for me, the device can not be detected.

* I have downloaded the VCP drivers: https://ftdichip.com/drivers/vcp-drivers/
* I have followed the instructions: https://slae.sh/projects/cc2652/#smart-rf-flash-programmer2
  * I assume the device needs to be plugged in while clicking the buttons as per the guide above
* ZigsStar can not detect the device as well

Koenkk, 2022-09-12:
Do you have another Windows pc where you can check on? Looks like an issue with the PC, for others it works fine.

boorac, 2022-10-03:
Took me a while, but I can confirm that I was able to get things up and working by using Flash programmer 2 and repeating the above steps, plus updating HA and all available extensions. Thanks for the hint!

closed 2022-10-03 by boorac