Koenkk/zigbee2mqtt #12954: OTA update of Hue Adore always returns old version

mshm8, 2022-06-26:
### What happened?

I'm trying to OTA update the Philips 3435011P7 (Hue white ambiance bathroom ceiling light Adore). It's currently on version 1.16.1_r19181. 
Zigbee2MQTT shows that an update is available. I can start it without any problems, it's running up to 100% and gives a success message. Light is flashing. But it's still on the same firmware version:
```
info  2022-06-26 21:55:05: Device '0x0017880104504b4d' was updated from '{"dateCode":"20160907","softwareBuildID":"1.16.1_r19181"}' to '{"dateCode":"20160907","softwareBuildID":"1.16.1_r19181"}'
```
``` 
info  2022-06-26 21:55:05: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":{"date_code":"20160907","software_build_id":"1.16.1_r19181"},"id":"0x0017880104504b4d","to":{"date_code":"20160907","software_build_id":"1.16.1_r19181"}},"status":"ok","transaction":"wn169-2"}'
```

I tried to repair the device several times and retried the update. Same result.

Thanks for any idea or help.

### What did you expect to happen?

OTA update to a newer version than 1.16.1_r19181. 

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus CP2102N

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8987412/log.txt)

Koenkk, 2022-06-27:
Probably this device requires a 2 step update, @[TheJulianJES](https://github.com/Koenkk/zigbee-OTA/commits?author=TheJulianJES) how can the images be retrieved?

TheJulianJES, 2022-06-27:
Image Type 270, version 16783620:
http://fds.dc1.philips.com/firmware/ZGB_100B_010E/16783620/100B-010E-01001904-ConfLight-ModuLum_0012.zigbee

Koenkk, 2022-06-27:
@mshm8 can you check if it work now?

closed 2022-06-27 by Koenkk

mshm8, 2022-06-27:
Yes, working like a charm now. After 2 steps of OTA it’s now on 1.88.1 (20210331).
Thank you so much guys!

Koenkk, 2022-06-28:
Awesome @TheJulianJES 😄