Koenkk/zigbee2mqtt #12955: [New device support]: GI:EXperience zigbee water valve

tbelaire, 2022-06-27:
### Link

https://www.aliexpress.com/item/1005004222098040.html?spm=a2g0o.productlist.0.0.55795d7409khpn&algo_pvid=1cfade68-8cc5-471c-8b3d-13c2649ce4c2&algo_exp_id=1cfade68-8cc5-471c-8b3d-13c2649ce4c2-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000028429008661%22%7D&pdp_npi=2%40dis%21CAD%21%2178.45%2170.6%21%21%21%21%40210318cf16562967990161683ef2b6%2112000028429008661%21sea

### Database entry

{"id":38,"type":"EndDevice","ieeeAddr":"0xa4c1387b3d12c3c4","nwkAddr":766,"manufId":4417,"manufName":"_TZE200_sh1btabb","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"/>K*i9>K*iC>K*iM>K*iW>K*ia>K*ik>K*iu>K*i>K*i�>K*i�>K*i\u0010>K*i","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1656286238642,"defaultSendRequestWhen":"immediate"}

### Comments

I started following the tuya new device guide
https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#_9-find-your-data-point

I have collected the following from playing with the device and watching the tuya iot developer logs.

```
Function         |      Code  |  Notes

Switch           |          1 | Not sure, less used than Start. ["On", "Off"]
Start            |          2 | Used to control the flow of water.  ["on", "off"]
上次灌溉时间       |        101 | Last Irrigation Record
下次灌溉时间       |        102 | Next Irrigation
实时灌溉方式(次数)  |        103 | Realtime irrigation method, number of times
时长              |        104 | (Duration, but maybe actually capacity)  15秒 (15 seconds == 15 Litres)
实时灌溉间隔(秒)    |        105 | Realtime interval (time interval in seconds)
当前温度(摄氏度)    |        106 | Current temperature (C)  (Seems to report 0 or 1)
Smart Weather    |         107 | "range": ["sunny","cloudy","rainy", "snowy"]
当前电量(%)        |        108 | Current battery (0% - 100%)
实时累计时长(秒)    |        110 | Cummulative realtime duration (seconds)
实时累计水量(升)    |        111 | Cummulative realtime capacity (Litres)
其它扩展            |       112 | Other extensions, a 1 was set once, not sure what it does.
定时功能            |       113 | Timer setting, haven't seen it used yet.
统计功能            |       114 | Statistics, 00:01:06,0  How long, and how much did it water.
```

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

//Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_sh1btabb'
// Received Zigbee message from '0xa4c1387b3d12c3c4', type 'commandDataReport', 
// cluster 'manuSpecificTuya', 
// data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":103}],
//        "seq":31488}' from endpoint 1 with groupID null
const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_sh1btabb'
        }
    ],
    model: 'QT-06', // Vendor model number, look on the device for a model number
    vendor: 'GIExperience', // Vendor of the device (only used for documentation and startup logging)
    description: 'Wireless water valve', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [
        fz.tuya_data_point_dump,
        fz.ignore_basic_report,
    ],
    toZigbee: [
        tz.tuya_data_point_test
    ],
    onEvent: tuya.onEventSetTime, // commandMcuSyncTime
    configure: async (device, coordinatorEndpoint, logger) => {
        logger.info("Configure ran");
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.battery()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

tbelaire, 2022-06-27:
I have a HA addon install, and I'm not sure how to go about modifying the `ode_modules/zigbee-herdsman-converters/converters/fromZigbee.js` files, I think the HA docker blocks me from seeing them.

Andy-Hennah, 2022-06-27:
See issue #12778 - there may be some useful information there to identify the codes.

Koenkk, 2022-06-27:
This device is already supported in the latest dev.  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

tbelaire, 2022-06-27:
Sweet, I'll switch to running dev.  And here I was about to duplicate the work.