Koenkk/zigbee2mqtt #12956: RGB Genie ZB-5001 Home Assistant Device Triggers are Only Partially Useful

scolby33, 2022-06-27:
### What happened?

The RGB Genie ZB-5001 remote has 5 scene on/off buttons. When it is paired to Zigbee2MQTT and discovered in Home Assistant using device discovery, it is impossible to differentiate between the scene buttons.

The following action topics are used:
- homeassistant/device_automation/0x<id>/action_on/config
- homeassistant/device_automation/0x<id>/action_off/config
- homeassistant/device_automation/0x<id>/action_brightness_move_up/config
- homeassistant/device_automation/0x<id>/action_brightness_stop/config
- homeassistant/device_automation/0x<id>/action_brightness_move_down/config

The action_on/config message, for example is like:
```yaml
automation_type: trigger
device:
  identifiers:
    - zigbee2mqtt_0x<id>
  manufacturer: RGB Genie
  model: Zigbee 3.0 remote control (ZB-5001)
  name: Living Room Remote
  sw_version: 2.5.3_r20
payload: 'on'
subtype: 'on'
topic: zigbee2mqtt/Pretty Name/action
type: action
platform: mqtt
```

Unfortunately, this results in the same device trigger for each button.

Zigbee2MQTT can obviously tell the buttons apart, using the `groupID` field, which is translated to the `action_group` payload value. This is the log of me pressing scene 1 on followed by scene 2 on:
```
Zigbee2MQTT:debug 2022-06-27 04:47:54: Received Zigbee message from 'Pretty Name', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 62465
Zigbee2MQTT:info  2022-06-27 04:47:54: MQTT publish: topic 'zigbee2mqtt/Pretty Name', payload '{"action":"on","action_group":62465,"battery":43.5,"linkquality":109}'
Zigbee2MQTT:info  2022-06-27 04:47:54: MQTT publish: topic 'zigbee2mqtt/Pretty Name/action', payload 'on'
Zigbee2MQTT:debug 2022-06-27 04:47:56: Received Zigbee message from 'Pretty Name', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 2 with groupID 62466
Zigbee2MQTT:info  2022-06-27 04:47:56: MQTT publish: topic 'zigbee2mqtt/Pretty Name', payload '{"action":"on","action_group":62466,"battery":43.5,"linkquality":109}'
Zigbee2MQTT:info  2022-06-27 04:47:56: MQTT publish: topic 'zigbee2mqtt/Pretty Name/action', payload 'on'
```

### What did you expect to happen?

Zigbee2MQTT would have discovery for Home Assistant with separate actions for each of the 5 scene on/off buttons, something like:
- homeassistant/device_automation/0x<id>/action_1_on/config
- homeassistant/device_automation/0x<id>/action_1_off/config
- homeassistant/device_automation/0x<id>/action_1_brightness_move_up/config
- homeassistant/device_automation/0x<id>/action_1_brightness_stop/config
- homeassistant/device_automation/0x<id>/action_1_brightness_move_down/config
- homeassistant/device_automation/0x<id>/action_2_on/config
- ...

### How to reproduce it (minimal and precise)

1. Pair RGB Genie ZB-5001 to Zigbee2MQTT
2. Press each of the buttons for Home Assistant discovery to happen
3. Observe that there is only one of each type of action and it is not possible to use device triggers to differentiate different buttons in Home Assistant.

### Zigbee2MQTT version

1.25.2 commit: 3c5854fa

### Adapter firmware version

20220219

### Adapter

Tube's CC2652P2 USB Coordinator

### Debug log

See relevant logs above in the "What happened?" section.

Koenkk, 2022-06-27:
This is a limitation of HA (you cannot respond to the full payload). You can use https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt instead

scolby33, 2022-06-27:
Yes, thank you for pointing out the direct MQTT option.

Is it not possible for Zigbee2MQTT to use separate topics for each groupID/action_group? (I would appreciate if you could differentiate between "not possible due to how Zigbee works" and "not possible because Zigbee2MQTT doesn't have the feature." If it's the latter, would you be open to a pull request? The device triggers are _so_ much more convenient than having to mange MQTT topic names manually 😊)

closed 2022-08-05 by github-actions[bot]