Koenkk/zigbee2mqtt #12957: RGB Genie ZB-5001 Master On/Off Buttons are not Handled Properly

scolby33, 2022-06-27:
### What happened?

The RGB Genie ZB-5001 remote has two "master" buttons at the top: one for on and one for off.

When one of these buttons are pressed, Zigbee messages are sent for _each_ of the device's action groups. However, only _one_ MQTT message is emitted by Zigbee2MQTT (from the same action group as scene 1).

(For some reason, the endpoint registered is always the same as the last individual scene button pressed. For example, if I press "scene 2 off," a Zigbee message is sent from endpoint 2 with groupID 62466. If I then press the master off, the 5 Zigbee messages (for each of groupID 6246{5, 6, 7, 8, 9}) are sent from endpoint 2. If I press "scene 1 on" then "master off," the messages will come from endpoint 1, etc.)

### What did you expect to happen?

An MQTT message for each of the incoming Zigbee messages would be sent.

### How to reproduce it (minimal and precise)

1. Pair RGB Genie ZB-5001 with Zigbee2MQTT
2. Press the master on or off button
3. Observe only 1 outgoing MQTT message

### Zigbee2MQTT version

1.25.2 commit: 3c5854f

### Adapter firmware version

20220219

### Adapter

Tube's CC2652P2 USB Coordinator

### Debug log

```
Zigbee2MQTT:debug 2022-06-27 04:55:29: Received Zigbee message from 'Pretty Name', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 2 with groupID 62465
Zigbee2MQTT:info  2022-06-27 04:55:29: MQTT publish: topic 'zigbee2mqtt/Pretty Name', payload '{"action":"on","action_group":62465,"battery":43.5,"linkquality":109}'
Zigbee2MQTT:info  2022-06-27 04:55:29: MQTT publish: topic 'zigbee2mqtt/Pretty Name/action', payload 'on'
Zigbee2MQTT:debug 2022-06-27 04:55:29: Received Zigbee message from 'Pretty Name', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 2 with groupID 62466
Zigbee2MQTT:debug 2022-06-27 04:55:29: Received Zigbee message from 'Pretty Name', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 2 with groupID 62467
Zigbee2MQTT:debug 2022-06-27 04:55:29: Received Zigbee message from 'Pretty Name', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 2 with groupID 62468
Zigbee2MQTT:debug 2022-06-27 04:55:29: Received Zigbee message from 'Pretty Name', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 2 with groupID 62469
```

Koenkk, 2022-06-27:
Should be fixed now!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

scolby33, 2022-06-28:
It works! Thank you!

closed 2022-06-28 by scolby33