Koenkk/zigbee2mqtt #12961: Low brightness only possible after higher brightness first

MPowerData, 2022-06-27:
### What happened?

Using the Eco-Dim.07 zigbee dimmer I want to put my lights on low brightness, namelijk value '33'. However, the lights won't turn on. 

First I have to set them to a higher value, for example '50', and afterwards I can change it to '33'. Then it works. 

Why do I need to set it to a higher value first? Is this something I can prevent.

### What did you expect to happen?

The lights turn on directly with the specific brightness value.

### How to reproduce it (minimal and precise)

Send brightness value 33, lights don't turn on.
Send brightness value 50, lights turn on.
Send brightness value 33, lights stay on with the correct value.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20210708

### Adapter

Slaesh cc2652rb stick

MPowerData, 2022-07-28:
Any idea on this?

Koenkk, 2022-07-28:
Could you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

MPowerData, 2022-08-16:
Sorry this took a while, but hereby the log:

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9349606/log.txt)

Koenkk, 2022-08-16:
Checked your logs and I see the exact same Zigbee commands are being send for 50 and 33 (expect the different brightness value of course). This indicates that this issue seems to be on the device side. Are you sure the connected bulb does support dimming correctly? What happens if you physically control the dimmer?

MPowerData, 2022-08-22:
Yes, there are 3 spots connected to the dimmer. Physically I can dim them to 28 even, going lower turns of the lights. I first need to go above 50 again to turn them on. So yeah, weird problem... :-)

closed 2022-09-29 by github-actions[bot]