Koenkk/zigbee2mqtt #12971: IKEA Tradfri Remote (E1810) only sending 'toggle' action

llamalump, 2022-06-28:
### What happened?

I tried asking this in the Discord but got no responses, so asking here in the hopes I get some assistance.

I have some IKEA Tradfri remotes (model number E1810) that I have hooked into Zigbee2mqtt.

Up until now I've just been using these remotes as a dumb dimmer switch (toggle on/off, brightness up/down), having bound all the lights I want them to turn on and off directly to the remotes in the Z2M interface (no Node-RED automations). It's been working fine so far, minus the odd hiccup of having to re-bind one of the lights to its respective remote every now and again. 

I've been meaning to set up some simple automations in Node-RED for some time, by leveraging the different action types that Zigbee2mqtt exposes for these remotes (https://www.zigbee2mqtt.io/devices/E1524_E1810.html#action-enum), namely `toggle_hold` (as described in https://github.com/Koenkk/zigbee2mqtt/issues/2077#issuecomment-538691885), but as I've just found out, the only action that these remotes are currently publishing for me is the `toggle` action.

There's some odd behaviour happening as well -- when I press and hold the on/off button (middle button), if I hold it long enough it will turn the lights on at full brightness (even if they were dimmed before).

Has anyone encountered this before? Is there a solution?

### What did you expect to happen?

The only action that these remotes are currently publishing for me is the `toggle` action. I would expect that the remotes work with the actions described in the Z2M documentation: https://www.zigbee2mqtt.io/devices/E1524_E1810.html#action-enum

### How to reproduce it (minimal and precise)

I'm not sure if it's a problem with my system alone, or if it's a wider problem with E1810 Tradfri remotes, but here goes anyway:

* Pair a Tradfri E1810 remote to Z2M
* Press any of the buttons
* Watch as none of the actions described in the documentation are published, except for `toggle`

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

0x26780700

### Adapter

ConBee II

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/8999415/log.txt)

Koenkk, 2022-06-28:
It might help to;

- force remove the device from via the z2m frontend
- stop z2m
- take out the remote battery; wait 30 seconds
- start z2m
- permit joining and repair the device

llamalump, 2022-06-30:
Thanks, that appears to have worked. Question is, why would it suddenly stop working properly and require re-pairing?

Koenkk, 2022-07-01:
I'm not sure yet what the root cause can be, but it may have something to do with replacing or almost empty batteries making the remote lose its config.

llamalump, 2022-07-03:
I don't think it's that; the remote reports having 87% battery remaining and I haven't removed/replaced the battery since I added it to the network months ago. I'll keep looking through the logs to see if there's anything revealing.

closed 2022-08-10 by github-actions[bot]

werdck, 2024-01-10:
For anyone finding this later on (via search, as I did for example):

The TRADFRI remotes also stop sending any action but "toggle" when they get bound to one or more devices.
This behavior affects at least firmware 24.4.5, where it only applies to devices, not groups.