Koenkk/zigbee2mqtt #12976: failed (no response received) - Conbee II

FabioEight, 2022-06-28:
### What happened?

Since today, without apparent reason, Im no longer able to interact with my zigbee switches and lights.
I’m using a Conbee 2 with the latest firmware and no changes were made during the last days.
Looking on the logs, I see this error when I try to change a state of a light:

failed (no response received)

im still able to retrieve data from temperature and motion sensors

### What did you expect to happen?

Interact with switches and ,ights

### How to reproduce it (minimal and precise)

Try to interact with a light or switch.
Tried to reboot and restart z2m addon but no luck.

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

0x26780700

### Adapter

Conbee II 

### Debug log

[Nota 28 giu 2022 20_55_37.txt](https://github.com/Koenkk/zigbee2mqtt/files/9003940/Nota.28.giu.2022.20_55_37.txt)

FabioEight, 2022-06-28:
Update: I tried to relink some devices but nothing has changed. 
seems that Z2M Is unable to change the state of the entities, while in some cases it receive the status update properly (ex. If a light is manually turned on).
Also, if I click 10x to toggle a switch, after a while the command is properly sent.

Tapematch, 2022-07-21:
I think I have the same problem. Also using a Conbee II Stick with a Raspberry Pi 4 running the latest Version as Home Assistant Addon. I alread restart zigbee2mqtt, the Host and all my lights. I also tried with the Edge Version, but no Change.
Since around 3 days suddenly all my lights dont work anymore, no Errors shown in the Log in Startup. Somehow almost all of my devices Show a LQI of 255?
Did you find a solution for this?

FabioEight, 2022-07-21:
> I think I have the same problem. Also using a Conbee II Stick with a Raspberry Pi 4 running the latest Version as Home Assistant Addon.
> Since around 3 days suddenly all my lights dont work anymore, no Errors shown in the Log in Startup.
> I already restarted zigbee2mqtt, the Host and all my lights. I also tried with the Edge Version, but no change.
> Somehow almost all of my devices show a LQI of 255?
> Did you find a solution for this?

I tried to upgrade the firmware but no luck. 
After almost a day of troubleshooting, I took the opportunity to migrate my devices to the newer Sonoff USB dongle

closed 2022-08-29 by github-actions[bot]

chris400, 2022-09-28:
I am also affected by this problem.

bonzini, 2022-11-11:
Duplicate of #11647.