Koenkk/zigbee2mqtt #12986: lumu.weather (Xiaomi WSDCGQ11LM) now has a `power_outage_count` attribute

sjorge, 2022-06-29:
### What happened?

I noticed today that the *lumi.weather* device now has a `power_outage_count` attribute, not sure if that even makes sense and the number show does not go up when removing and inserting the battery so it's probably incorrect.

### What did you expect to happen?

It not to have this attribute as it also makes no sense for the temp/hum/pressure sensor.

### How to reproduce it (minimal and precise)

Just let the device be paired for a day or so and it shows up.

### Zigbee2MQTT version

1.25.2-dev (3d406c3c)

### Adapter firmware version

20220219

### Adapter

zzh

sjorge, 2022-06-29:
I managed to capture it just now as I was writing this report

```
Zigbee2MQTT:debug 2022-06-29 20:24:02: Received Zigbee message from '0x00158d0003f115c5', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3135,"10":0,"100":2785,"101":5475,"102":100792,"4":5032,"5":83,"6":[0,2]}}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-06-29 20:24:02: lumi.weather: unknown key 6 with value 0,2
Zigbee2MQTT:debug 2022-06-29 20:24:02: lumi.weather: Processed data into payload {"voltage":3135,"battery":100,"power_outage_count":82,"temperature":27.85,"humidity":54.75,"pressure":1007.9}
Zigbee2MQTT:debug 2022-06-29 20:24:02: lumi.weather: Processed data into payload {"voltage":3135,"battery":100,"power_outage_count":82,"temperature":27.85,"humidity":54.75,"pressure":1007.9}
Zigbee2MQTT:info  2022-06-29 20:24:02: MQTT publish: topic 'zigbee2mqtt/0x00158d0003f115c5', payload '{"battery":100,"humidity":54.75,"linkquality":138,"power_outage_count":82,"pressure":1007.9,"temperature":27.85,"voltage":3135}'
```

sjorge, 2022-06-29:
I think it got added 4 months ago by this commit https://github.com/Koenkk/zigbee-herdsman-converters/commit/2554e6d1673b4d315f07a1562a7b95d1fe2d05d8

I don't generally use this sensor I was just using it while working on a feature I want to add to zigbee2mqtt as a test device on my test network. So I guess it's been there for a while.

sjorge, 2022-06-29:
OK, back to what I was actually trying to accomplish :)

McGiverGim, 2022-06-30:
Not really, it was added here:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4362

closed 2022-06-30 by sjorge