Koenkk/zigbee2mqtt #12997: No Update Firmware Error

maxbec, 2022-06-30:
### What happened?

Zigbee2Mqtt is showing that there is no update available for a Philipps Hue Bar. For the other with the exact same Product ID there was a succesful update.
[image]

### What did you expect to happen?

Showing that there is a new update available.

### How to reproduce it (minimal and precise)

No Idea.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Plus Dongle

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9020401/log.txt)

maxbec, 2022-07-10:
Any idea?

Koenkk, 2022-07-11:
Maybe the cached info at the z2m side is incorrect. Update it by doing a read on the following attributes via the z2m frontend and restart z2m:

[image]

maxbec, 2022-08-12:
> Maybe the cached info at the z2m side is incorrect. Update it by doing a read on the following attributes via the z2m frontend and restart z2m:
> 
> [image]

Done! But it still shows the old version:
[image]

Three same bulbs with everyone on a different version. Check for update brings no update!
[image]

closed 2022-09-20 by github-actions[bot]

maxbec, 2022-11-09:
Can you give an update here?