Koenkk/zigbee2mqtt #12999: ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY

robinemde, 2022-07-01:
### What happened?

Network Key in configuration.yaml was changed to a wrong value to test setting it via env var.
Correct Network Key provided to the Z2M Docker Container via ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY.

### What did you expect to happen?

Network Key setting in configuration.yaml should be overwritten by the environment variable.

Changing back to correct key in configuration.yaml is not sufficient, I also have to remove the environment variable to be passed to the container, so it seems it is used somehow.

I have tried different syntaxes like

ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY=[0x1a,0x2b,0x3c,0x4d,0x5e,0x6f,0x7a,0x8b,0x9c,0x0d,0x1e,0x2f,0x3a,0x4b,0x5c,0x6d]

(just an example key, with/without space, with/without " or ' for encapsulation, etc.)

### How to reproduce it (minimal and precise)

Generated Network Key, paired device to check everything is functional.

Copy correct Network Key from configuration.yaml to env var ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY.
Set Network Key in configuration.yaml to a wrong value.

Used env var ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY and passed it to Z2M Docker Container.

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 Dongle

### Debug log

zigbee2mqtt  | SyntaxError: Unexpected token x in JSON at position 2
zigbee2mqtt  |     at JSON.parse (<anonymous>)
zigbee2mqtt  |     at /app/lib/util/settings.ts:376:49
zigbee2mqtt  |     at Array.forEach (<anonymous>)
zigbee2mqtt  |     at iterate (/app/lib/util/settings.ts:361:26)
zigbee2mqtt  |     at /app/lib/util/settings.ts:396:21
zigbee2mqtt  |     at Array.forEach (<anonymous>)
zigbee2mqtt  |     at iterate (/app/lib/util/settings.ts:361:26)
zigbee2mqtt  |     at /app/lib/util/settings.ts:396:21
zigbee2mqtt  |     at Array.forEach (<anonymous>)
zigbee2mqtt  |     at iterate (/app/lib/util/settings.ts:361:26)

robinemde, 2022-07-01:
Hello again,

and played around with different changes and found that the following is working:

ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY="[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16]"

I converted the hex values to decimal. So, it seems Hex is not supported, but in the configuration.yaml an array of hex values is working. Only env var seems not to be parsed/converted properly.

Koenkk, 2022-07-01:
I think this has been fixed in #12948 which is available in z2m 1.26.0

Sheng-IoT, 2022-07-07:
I had the same issue, and here is my docker-compose.yml:

```
version: '3.8'
services:
  mqtt:
    image: eclipse-mosquitto:2.0
    restart: unless-stopped
    volumes:
      - "./mosquitto-data:/mosquitto"
    ports:
      - "1883:1883"
      - "9001:9001"
    command: "mosquitto -c /mosquitto-no-auth.conf"

  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt:1.26.0
    volumes:
      - ./data:/app/data
      #- ./zigbee2mqtt-data:/app/data
      #- /run/udev:/run/udev:ro
    ports:
      - 8002:8080
    environment:
      - TZ=Europe/Berlin
      - ZIGBEE2MQTT_CONFIG_PERMIT_JOIN=true
      - ZIGBEE2MQTT_CONFIG_MQTT_SERVER=mqtt://mqtt
      - ZIGBEE2MQTT_CONFIG_MQTT_BASE_TOPIC=zigbee2mqtt
      - ZIGBEE2MQTT_CONFIG_SERIAL_PORT=/dev/ttyUSB0
      - ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY=GENERATE
      - ZIGBEE2MQTT_CONFIG_FRONTEND=true
    devices:
      - /dev/ttyUSB0:/dev/ttyUSB0

```
as u see, I set `- ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY=GENERATE` because it will "Let Zigbee2MQTT generate a new network key on first start", but when I do docker-compose up, I got the following error:

```
$ docker-compose up
[+] Running 1/1
 ⠿ zigbee2mqtt Pulled                                                                                              3.7s
[+] Running 3/3
 ⠿ Network zigbee2mqtt_default   Created                                                                           0.1s
 ⠿ Container zigbee2mqtt-mqtt-1  Created                                                                           0.2s
 ⠿ Container zigbee2mqtt         Created                                                                           0.2s
Attaching to zigbee2mqtt, zigbee2mqtt-mqtt-1
zigbee2mqtt-mqtt-1  | 1657167233: mosquitto version 2.0.14 starting
zigbee2mqtt-mqtt-1  | 1657167233: Config loaded from /mosquitto-no-auth.conf.
zigbee2mqtt-mqtt-1  | 1657167233: Opening ipv4 listen socket on port 1883.
zigbee2mqtt-mqtt-1  | 1657167233: Opening ipv6 listen socket on port 1883.
zigbee2mqtt-mqtt-1  | 1657167233: mosquitto version 2.0.14 running
zigbee2mqtt         | Using '/app/data' as data directory
zigbee2mqtt         | Creating configuration file...
zigbee2mqtt         | SyntaxError: Unexpected token G in JSON at position 0
zigbee2mqtt         |     at JSON.parse (<anonymous>)
zigbee2mqtt         |     at /app/lib/util/settings.ts:384:49
zigbee2mqtt         |     at Array.forEach (<anonymous>)
zigbee2mqtt         |     at iterate (/app/lib/util/settings.ts:369:26)
zigbee2mqtt         |     at /app/lib/util/settings.ts:404:21
zigbee2mqtt         |     at Array.forEach (<anonymous>)
zigbee2mqtt         |     at iterate (/app/lib/util/settings.ts:369:26)
zigbee2mqtt         |     at /app/lib/util/settings.ts:404:21
zigbee2mqtt         |     at Array.forEach (<anonymous>)
zigbee2mqtt         |     at iterate (/app/lib/util/settings.ts:369:26)
zigbee2mqtt exited with code 1
zigbee2mqtt         | SyntaxError: Unexpected token G in JSON at position 0
zigbee2mqtt         |     at JSON.parse (<anonymous>)
zigbee2mqtt         |     at /app/lib/util/settings.ts:384:49
zigbee2mqtt         |     at Array.forEach (<anonymous>)
zigbee2mqtt         |     at iterate (/app/lib/util/settings.ts:369:26)
zigbee2mqtt         |     at /app/lib/util/settings.ts:404:21
zigbee2mqtt         |     at Array.forEach (<anonymous>)
zigbee2mqtt         |     at iterate (/app/lib/util/settings.ts:369:26)
zigbee2mqtt         |     at /app/lib/util/settings.ts:404:21
zigbee2mqtt         |     at Array.forEach (<anonymous>)
zigbee2mqtt         |     at iterate (/app/lib/util/settings.ts:369:26)
zigbee2mqtt exited with code 1
```
It seems the value of **ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY** is parsed in the wrong way.

Koenkk, 2022-07-07:
You should not set always `ZIGBEE2MQTT_CONFIG_ADVANCED_NETWORK_KEY` to `GENERATE`, it will generate a new network key on **every** start (thus throwing away your old network everytime). The use case of `GENERATE` is to set it once in the `configuration.yaml`, on start z2m will update `configuration.yaml` with the actual network key.

robinemde, 2022-07-07:
Sure, that is absolutely clear, but I would like to be able to trigger it by ENVIRONMENT VAR, let it be written to configuration.yaml and then copy it from there to the ENV VAR.

Koenkk, 2022-07-08:
I see, fixed the issue now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-15 by github-actions[bot]