Koenkk/zigbee2mqtt #13001: MQTT failed to connect: Connection refused: Not authorized

henrikrox, 2022-07-01:
### What happened?

After updating from 1.24.0-1 to 1.25.2-1

it wont start up. 

Logs says 

```
Zigbee2MQTT:info  2022-07-01 16:12:46: Enocean Bryter Stue (0x0000000001722ff9): PTM 215Z - EnOcean Pushbutton transmitter module (GreenPower)
Zigbee2MQTT:info  2022-07-01 16:12:46: Varmepumpe kjeller (0x0015bc002f007453): SPLZB-131 - Develco Power plug (Router)
Zigbee2MQTT:info  2022-07-01 16:12:46: Spottskinne Stue3 (0x50325ffffec97bd7): LED1923R5 - IKEA TRADFRI LED bulb GU10 345 lumen, dimmable, white spectrum, color spectrum (Router)
Zigbee2MQTT:info  2022-07-01 16:12:46: spottskinne soverom 1 (0x14b457fffe4b7c30): LED1842G3 - IKEA TRADFRI LED bulb E27 WW clear 250 lumen, dimmable (Router)
Zigbee2MQTT:info  2022-07-01 16:12:46: Sportskinne soverom 2 (0x14b457fffe4b75a3): LED1842G3 - IKEA TRADFRI LED bulb E27 WW clear 250 lumen, dimmable (Router)
Zigbee2MQTT:info  2022-07-01 16:12:46: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-07-01 16:12:46: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:error 2022-07-01 16:12:46: MQTT failed to connect: Connection refused: Not authorized
Zigbee2MQTT:error 2022-07-01 16:12:46: Exiting...
Zigbee2MQTT:info  2022-07-01 16:12:46: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-07-01 16:12:47: Stopped zigbee-herdsman
[16:12:48] INFO: Handing over control to Zigbee2mqtt Core ...
```

I havnt changed any password etc, what can i do to fix this?

### Zigbee2MQTT version

1.25.2-1

### Adapter firmware version

cantlogintofrontend

### Adapter

deconz

Koenkk, 2022-07-01:
Can you confirm the correct config is in the HA addon MQTT section?

sim-san, 2022-08-08:
I got the same error after upgrading from 1.26.0-1 to 1.27.0-1. After a complete host reboot, the connection works for some hours.

I use an secret on both addons: [core-mosquitto](https://github.com/home-assistant/addons/tree/master/mosquitto) and [hassio-zigbee2mqtt](https://github.com/zigbee2mqtt/hassio-zigbee2mqtt)

sim-san, 2022-08-08:
here some logs:
<details>
  <summary>HomeAssitant Addon Log</summary>
  
[16:16:47] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Zigbee2MQTT:info  2022-08-08 16:16:54: Logging to console and directory: '/config/zigbee2mqtt/log/2022-08-08.16-16-54' filename: log.txt
Zigbee2MQTT:info  2022-08-08 16:16:54: Starting Zigbee2MQTT version 1.27.0 (commit #unknown)
Zigbee2MQTT:info  2022-08-08 16:16:54: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:info  2022-08-08 16:17:05: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-08-08 16:17:05: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-08-08 16:17:05: Currently 5 devices are joined:
Zigbee2MQTT:info  2022-08-08 16:17:05: EckenlampeOben (0x00158d000341e87f): 404000/404005/404012/404019 - Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, color, opal white (Router)
Zigbee2MQTT:info  2022-08-08 16:17:05: ikea_5way_remote (0xbc33acfffe8707f3): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
Zigbee2MQTT:info  2022-08-08 16:17:05: WakeUpLicht (0x00158d000341f04d): 404000/404005/404012/404019 - Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, color, opal white (Router)
Zigbee2MQTT:info  2022-08-08 16:17:05: aqara_kinder (0x00158d000320d2d3): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2022-08-08 16:17:05: aqara_eltern (0x00158d000678f441): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2022-08-08 16:17:05: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-08-08 16:17:05: Connecting to MQTT server at mqtt://core-mosquitto
Zigbee2MQTT:error 2022-08-08 16:17:06: MQTT failed to connect: Connection refused: Not authorized
Zigbee2MQTT:error 2022-08-08 16:17:06: Exiting...
Zigbee2MQTT:info  2022-08-08 16:17:06: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-08-08 16:17:15: Stopped zigbee-herdsman
[16:17:16] INFO: Handing over control to Zigbee2MQTT Core ...
> zigbee2mqtt@1.27.0 start
> node index.js
Zigbee2MQTT:info  2022-08-08 16:17:23: Logging to console and directory: '/config/zigbee2mqtt/log/2022-08-08.16-17-23' filename: log.txt
Zigbee2MQTT:info  2022-08-08 16:17:23: Starting Zigbee2MQTT version 1.27.0 (commit #unknown)
Zigbee2MQTT:info  2022-08-08 16:17:23: Starting zigbee-herdsman (0.14.46)
Zigbee2MQTT:info  2022-08-08 16:17:35: zigbee-herdsman started (resumed)
Zigbee2MQTT:info  2022-08-08 16:17:35: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2022-08-08 16:17:35: Currently 5 devices are joined:
Zigbee2MQTT:info  2022-08-08 16:17:35: EckenlampeOben (0x00158d000341e87f): 404000/404005/404012/404019 - Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, color, opal white (Router)
Zigbee2MQTT:info  2022-08-08 16:17:35: ikea_5way_remote (0xbc33acfffe8707f3): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
Zigbee2MQTT:info  2022-08-08 16:17:35: WakeUpLicht (0x00158d000341f04d): 404000/404005/404012/404019 - Müller Licht Tint LED bulb GU10/E14/E27 350/470/806 lumen, dimmable, color, opal white (Router)
Zigbee2MQTT:info  2022-08-08 16:17:35: aqara_kinder (0x00158d000320d2d3): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2022-08-08 16:17:35: aqara_eltern (0x00158d000678f441): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
Zigbee2MQTT:info  2022-08-08 16:17:35: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-08-08 16:17:35: Connecting to MQTT server at mqtt://core-mosquitto
Zigbee2MQTT:error 2022-08-08 16:17:35: MQTT failed to connect: Connection refused: Not authorized
Zigbee2MQTT:error 2022-08-08 16:17:35: Exiting...
Zigbee2MQTT:info  2022-08-08 16:17:35: Stopping zigbee-herdsman...
Zigbee2MQTT:info  2022-08-08 16:17:45: Stopped zigbee-herdsman
</details>

sim-san, 2022-08-08:
I try to add the port. But this also does not solve the problem:

> Zigbee2MQTT:info  2022-08-08 16:23:27: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2022-08-08 16:23:27: Connecting to MQTT server at mqtt://core-mosquitto:1883
Zigbee2MQTT:error 2022-08-08 16:23:27: MQTT failed to connect: Connection refused: Not authorized
Zigbee2MQTT:error 2022-08-08 16:23:27: Exiting...
Zigbee2MQTT:info  2022-08-08 16:23:27: Stopping zigbee-herdsman...

dankocrnkovic, 2022-08-11:
I had this problem today after restarting my host. 
Z2M could not connect to mqtt. Mosquitto broker was started and did not report any system errors but log had messages about unauthorized client.
I restarted host but problem was the same.

Then I stoped Z2M service and Mosquitto broker. Started Mosquitto broker and after a minute Z2M and then it was ok.

This is the first time that this happened in 2 years.

KBlansit, 2022-08-13:
I can confirm @dankocrnkovic. I had to restart my host system, and I got a "502: bad gateway" in the web portal error message along with the "MQTT failed to connect: Connection refused: Not authorized" in the log for zigbee2mqtt.

Following his procedure seemed to fix it for me.

For reference, I am running Zigbee2mqtt: 1.27.0-1 and Mosquitto broker: 6.1.2

sim-san, 2022-08-19:
> > I got the same error after upgrading from 1.26.0-1 to 1.27.0-1. After a complete host reboot, the connection works for some hours.
> 
> 
> 
> Can you provide the debug log until it starts failing?

How can I extract the debug log ?
I am using the Home Assistant Addon

lazarvgd, 2022-08-23:
This didn't work for me.
Is there a way to downgrade plugin version to 1.23 or 1.22?

juanchula, 2022-08-23:
I have the same problem with both version old (I think 1.22) and 1.27.
In my case I run zigbee2mqtt in docker and mosquito in HA. The only way I can solve it is by restarting the mosquito addon but every time I restart the whole system it happens again

[image]

sim-san, 2022-08-24:
I have to execute the same workaround with every restart

Koenkk, 2022-08-24:
> > > I got the same error after upgrading from 1.26.0-1 to 1.27.0-1. After a complete host reboot, the connection works for some hours.
> > 
> > 
> > Can you provide the debug log until it starts failing?
> 
> How can I extract the debug log ? I am using the Home Assistant Addon

For the HA addon it is in the `zigbee2mqtt/data/log` folder (you can use e.g. the HA file editor addon)

lunzai, 2022-08-28:
I'm facing the same issue, from time to time zigbee2mqtt will failed to start due to MQTT failed to connect: Connection refused: Not authorized
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438917/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438918/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438919/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438921/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438923/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438920/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438922/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438924/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9438925/log.txt)

Happened this morning.

AlfredJKwack, 2022-08-29:
> I have the same problem with both version old (I think 1.22) and 1.27. In my case I run zigbee2mqtt in docker and mosquito in HA. The only way I can solve it is by restarting the mosquito addon but every time I restart the whole system it happens again

> I have to execute the same workaround with every restart

Are you certain that mosquito is starting up properly after HA is restarted? There's an issue in _https://github.com/home-assistant/addons/issues/2618#issuecomment-1221296174_ alluding to a startup issue if you have mosquito set to _Start on boot_.

juanchula, 2022-09-03:
> > I have the same problem with both version old (I think 1.22) and 1.27. In my case I run zigbee2mqtt in docker and mosquito in HA. The only way I can solve it is by restarting the mosquito addon but every time I restart the whole system it happens again
> 
> > I have to execute the same workaround with every restart
> 
> Are you certain that mosquito is starting up properly after HA is restarted? There's an issue in _[home-assistant/addons#2618 (comment)](https://github.com/home-assistant/addons/issues/2618#issuecomment-1221296174)_ alluding to a startup issue if you have mosquito set to _Start on boot_.

Thanks.  That seemed to be the problem.  I disabled Start on boot and left Watchdog enabled and the problem did not occur again

raddy77, 2022-09-16:
> I had this problem today after restarting my host. Z2M could not connect to mqtt. Mosquitto broker was started and did not report any system errors but log had messages about unauthorized client. I [...]

Confirmed - this just happened to me as well after a restart because of a Zigbee dongle firmware update.
Is as if the MQTT broker does not get the proper login info from HA on startup (as I could not login with the Z2M user from the MQTT Explorer either).
Guess will have to disable the start on boot as well...

AlfredJKwack, 2022-09-16:
If you leave the watchdog on and turn off the start on boot you'll be fine and it won't require intervention. Quite counter-intuitive but it works. Mosquito will start up.

andrazek, 2022-10-02:
I can confirm the same issue occurred yesterday. Nothing else but restarting the Mosquitto broker add-on and then starting Z2M solved the issue. Before it worked fine from the beginning.