Koenkk/zigbee2mqtt #13002: [New device support]: smoke detector Zigbee model 'TS0601' and manufacturer name '_TZE200_ntcy3xu1'

heuvep, 2022-07-01:
### Link

smoke detector

### Database entry

Zigbee2MQTT:warn  2022-07-01 16:23:28: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ntcy3xu1'

### Comments

 2022-07-01 16:18:56Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ntcy3xu1'
Warning 2022-07-01 16:18:56Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### External converter

```shell
Warning 2022-07-01 16:20:05Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ntcy3xu1'
Warning 2022-07-01 16:20:05Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.htm
```

SamMousa, 2022-07-02:
this is a duplicate... there's already 2 other issues for the exact same device, they even include external converters.

Please make sure you at least search the issues for your device: https://github.com/Koenkk/zigbee2mqtt/issues?q=is%3Aissue+is%3Aopen+_TZE200_ntcy3xu1

Koenkk, 2022-07-02:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/12622 indeed

closed 2022-07-02 by Koenkk

heuvep, 2022-10-11:
Hello,

 

I am new at this but i did create the file and copied the information in it as shown. Also create the external converter but still no result afetr restart

 

Gt

 

Pierre