Koenkk/zigbee2mqtt #13011: SSM-U01  no power and energy readings since 1.23.0

TRIROG, 2022-07-01:
### What happened?

Since version 1.23.0 upwards no power or energy consumption data is being received from SSM-U01 devices.
I have tried to repair the devices by simply holding button for 8 seocnds and also removing the devices and then repairing them but nothing helped.

### What did you expect to happen?

Power and energy consumption is received if i rollback to 1.22.2, but in 1.23.0 and 1.24.0 

### Zigbee2MQTT version

1.24.0

### Adapter firmware version

0x26680700

### Adapter

ConBee2/RaspBee2

d3relict, 2022-07-08:
I have a few of these switches (7 in the network, 3 more in the drawer), and 2 of them lost their capability to report power. they all were recently repaired, some are new additions. it happened from one day to another, no change to any of the components, i just don't get any updates from some.

@Koenkk please advise! :)

Koenkk, 2022-07-08:
@TRIROG try to update your conbee firmware to [0x26720700](http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26720700.bin.GCF) or greater and repair the devices

@d3relict try with the latest z2m (1.26.0) and repair your device after upgrading z2m.

Koenkk, 2022-07-09:
@d3relict if they work no further steps are needed.

TRIROG, 2022-07-10:
Fixed. 
Conbee firmware update to 0x26780700 and repairing devices seems to fix the matter.

@Koenkk thank you for fast response and support!

closed 2022-07-10 by TRIROG