Koenkk/zigbee2mqtt #13012: Removed support for Phillips hue Modellnummer	‎929002269101?

Fernien, 2022-07-01:
### What happened?

I had the phillips hue 929002269101 working but since some update (dont know which since i didnt use the light for a long time) i cant controll it anymore. It says unrecognized device.
Did you remove the support for these?

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20201010

### Adapter

zstack

Koenkk, 2022-07-01:
Can you provide the data/database.db entry of your device?

Fernien, 2022-07-02:
How do i get it?

Fernien, 2022-07-02:
> data/database.db

How do i get this file? Can't find it

Koenkk, 2022-07-02:
Can you make a screenshot of your z2m frontend -> device -> about page?

Fernien, 2022-07-02:
> Can you make a screenshot of your z2m frontend -> device -> about page?

Is this the right one?

[image]

Koenkk, 2022-07-02:
Somehow the zigbee model is missing, can you try to remove the device (red bin button), restart z2m and repair the device?

Fernien, 2022-07-02:
I removed the device and for some reason z2m isnt starting anymore. Didn't change anything in the configuration.
Here's the log:

`Zigbee2MQTT:error 2022-07-02 21:08:02: Exiting...
Zigbee2MQTT:error 2022-07-02 21:08:02: TypeError: Cannot read properties of undefined (reading 'payload')
    at AdapterNvMemory.deleteItem (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/adapter-nv-memory.ts:141:39)
    at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:330:9)
    at ZnpAdapterManager.beginRestore (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:296:9)
    at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:80:17)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:123:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)`

closed 2022-08-10 by github-actions[bot]