Koenkk/zigbee2mqtt #13019: Xiaomi SJCGQ11LM Temperature incorrect

umtauscher, 2022-07-02:
### What happened?

Today I updated to 1.26. Since the change in 1.25.2 the temperature reading of the above mentioed sensor is to high by 5 °C 
I positioned this sensor in a refrigerator and it shows 13 °C instead of 8 °C which would be the correct temperature.

I realize, that the name of the tempertaur sensor has also been changed, but I suppose this device has an additinal sensor apart from the chip temeperature. I am using zigbee2mqtt with HA and restoring the 1.25.1-2 Version also restores the correct temperature reading.

### What did you expect to happen?

... that the temperature  reading (value) would not change after an update 

### How to reproduce it (minimal and precise)

1.25.1-2 correct temperature
1.25.2-1 wrong temperature
1.26 wrong temperature

### Zigbee2MQTT version

1.26

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0

ashokaiyar, 2022-07-02:
Since 1.25.2, the temperature exposed by this device is device_temperature. So you are looking at a stale value for "temperature" with 1.25.2 and 1.26.0 that is left in state.json by 1.25.1. The only accurate value is "device_temperature" which reflects the on-board chip temperature and not the climate temperature.

You can remove the stale values for "temperature" by manually editing (zigbee2mqtt-root)/data/state.json

umtauscher, 2022-07-02:
No I am not.  I am not talking about device temperature versus temperature.
Device temperature is 13 °C 
Temperature was 8 °C at the same time.
The higher temperature 1.26 shows is definately wrong.
I think you misunderstood that. I was not looking at a stale value.
When I use 1.26 the "device temnparature" shows 13 °C
When I switch back to 1.25.1-2 the "temperature" is 8 °C
That can't be right.

Koenkk, 2022-07-02:
The `temperature` will not be updated anymore since we renamed it to `device_temperature`, should you manually remove `temperature` from the `data/state.json` file.

umtauscher, 2022-07-02:
I really seem to have difficulties descibing the problem:
It's not the rename of the value, the problem is that the value (measurement) is wrong.

Koenkk, 2022-07-02:
@McGiverGim could the converter refactor have changed the temperature value?

McGiverGim, 2022-07-02:
No. I doubt that. The refactor was older if I'm not wrong.
Maybe it has configured some calibration of -5 for temperature? It will not work of course for device_temperature.
A debug log will help with this.

McGiverGim, 2022-07-03:
I've been looking at the code and I don't see anything that can produce 5 degrees of diference between versions. Only thing is a calibration of the value, that is a different attribute. @umtauscher can you confirm if you have a calibration applied to the temperature expose?
In the specific settings, compare the two versions, the one that work and the one that doesn't:
[image]

umtauscher, 2022-07-03:
Yes, that was it. There was an offset set in the old version that wasn't set in the new one that I wasn't aware of.
I'm sory guys for the false conclusion.
Thanks very much vor your help.
Cheers

closed 2022-07-04 by Koenkk