Koenkk/zigbee2mqtt #13020: Woox R7060 Garden irrigation no battery level/voltage (following)

jymorel, 2022-07-02:
### What happened?

Hi,
See also closed Issue #12163. 
Normaly solved in version 1.26.0-1, but I still have the same issue after upgrading. 
I have deleted the device then reappaired it.
But battery level is not displayed.
Furthermore, not sure that removing the voltage_level in this version was a good idea as it is useful information.

Warning 2022-07-02 10:30:04zigbee-herdsman-converters:WooxR7060: Unrecognized DP #107 with data {"dp":107,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Warning 2022-07-02 10:30:04zigbee-herdsman-converters:WooxR7060: Unrecognized DP #106 with data {"dp":106,"datatype":3,"data":{"type":"Buffer","data":[48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48]}}

### What did you expect to happen?

correct display of bateery (and voltage) display

### Zigbee2MQTT version

1.26.0-1

### Adapter firmware version

20220601.0

### Adapter

sonoff zigbee dongle plus

tabnul, 2022-07-11:
same issue here. upon joining i see errors as well, could be related.
have this issue on both z2m 1.22.2 and 1.16.0

```
Warning 2022-07-11 21:25:01zigbee-herdsman-converters:WooxR7060: Unrecognized DP #107 with data {"dp":107,"datatype":4,"data":{"type":"Buffer","data":[0]}}
Warning 2022-07-11 21:25:01zigbee-herdsman-converters:WooxR7060: Unrecognized DP #106 with data {"dp":106,"datatype":3,"data":{"type":"Buffer","data":[48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48,48]}}

```

ytzelf, 2022-07-11:
Similar issue here

Koenkk, 2022-07-12:
To find out what these data points mean see https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

rufik, 2022-07-26:
Same issue here :/

closed 2022-09-03 by github-actions[bot]