Koenkk/zigbee2mqtt #13025: [Feature request]: want device_options but applied to all devices of same model

sjorge, 2022-07-03:
### Is your feature request related to a problem? Please describe

Currently we can set a lot of configuration on the device entry in the config, or something that applies to everything by using the top level device_options.

However I've been finding my self wanting to apply some of those to entire sets of devices of the same model.
e.g. all Ubisys S1's or all Frient AirQuality sensors.

Searching around this was also brought up (by me 😂 ) a while ago here https://github.com/Koenkk/zigbee2mqtt/issues/9635#issuecomment-997193984

### Describe the solution you'd like

A new configuration option that has an additional layer something like:
```yaml
device_options_match:
  "vendor=SONOFF AND model=SNZB-02":
    debounce: 5
  "vendor=Philips OR vendor=IKEA":
    debounce: 1
    filtered_optimistic:
      - brightness
```

I'm not exactly sure how the matching would look, I prefer some flexibility as I feel I will regert it if I implement it say fixed match on model so having some login on at least `model` and `vendor` seems desirable.

I also kind of want to be able to filter on zigbeeModel, although for TuYa devices that gets annoying.

Another idea I had was seeing of I could re-use the fingerprint logic, which actually would be the most flexible but might make it a bit harder to use for other people.

So that would be something like

```yaml
device_options_fingerprint:
  - match:
     - type: EndDevice
       manufacturerName: eWeLink
       modelID: WB01
      endpoints:
        - ID: 1
          profileID: 260
          deviceID: 770
          inputClusters:
          - 0
          - 3
         - 1026
         - 1029
         - 1
        outputClusters:
        - 3
     - type: EndDevice
       manufacturerName: eWeLink
       modelID: TH01
    options:
      debounce: 5
  - match:
    - vendor: ['ikea', 'philips']
    options:
      debounce: 5
```

I'm not really super happy with either. I'll try to find some time to come up with something though but it might take a while as I try different things out.

### Describe alternatives you've considered

Dumping it all in the global one, which works unless 2 device models need different config for the same option.

### Additional context

n/a

sjorge, 2022-07-19:
I've been thinking about this on my daily walks and I don't like either of the above approaches, but think a hybrid might work.

```json
device_options_match:
  - match:
      vendor: SONOFF
      model: ['SNZB-02', 'SNZB-01']
    options:
      debounce: 1
  - match:
      vendor: !ikea
      type: EndDevice
    options:
      debounce: 5
```

So a array of objects, and each objects need a match and options object.

The options object is, well just the options to merge in. The match block can have a set of *properties* to match on, not sure how *extensive* these matchable properties should be but probably having at least: vendor, model, manufacturerName, modelID, and type seem like a good set.

Some basic 'logic' support for those properties seem like a minimal requirement to make this flexible enough are:
- OR logic ( provide a list e.g. `vendor: SONOFF` is just match on SONOFF but `vendor: ['SONOFF', 'IKEA']` will match SONOFF OR IKEA)
- invert logic (prepend !, e.g. `vendor: !SONOFF` to match every vendor that is NOT SONOFF)

I'll see if I can whip something up once it's less hot cause the heat is not really putting me in a programming mood.

Koenkk, 2022-07-19:
That looks quite nice. 

I would prefer to not separate this from the existing `device_options`. Maybe something like:

```yaml
device_options:
  retain: true
  debounce:
    match:
      vendor: SONOFF
      model: ['SNZB-02', 'SNZB-01']
    value: 1
  occupancy_timeout: 120
```

Implementation wise; this can nicely be added to [`device.ts`](https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/model/device.ts) (e.g. a method name `options()`)

sjorge, 2022-07-19:
I was thinking about that too, but at least looking at my config I want to set multiple stuff per device (talking about a ‘device’ as in all devices of the same kind), repeating the march for every properties makes it harder to read as you’re going to repeat the same march multiple times then and that makes updating them a pain.

let me think about this some more

sjorge, 2022-07-19:
I guess to fix part of the problem I'm still having is that `match` should really be `matches` and maybe that should be a list of match objects 🤔 

That still doesn't allow us to integrate it cleanly in the existing `device_options` though you still have the `match a bunch of devices` -> set of options to apply instead of the otherway around.

I think the problem is fundamentally that I want to match a set of devices that are described by there 'metadata' and apply a set of options.

I guess apply this option value to this set of devices is another way of looking at it, but that seems very backwards in my head.

As I always thing in 'sets of devices of the same kind'.

Koenkk, 2022-07-19:
What if we do:

```yaml
device_options:
  legacy: false
  debounce: 5
  qos: 2
  match:
      - vendor: IKEA
        model: ['E1524/E1810',  'E2001/E2002', 'E1743']
        options:
          debounce: 0
```

sjorge, 2022-07-19:
> What if we do:
> 
> ```yaml
> device_options:
>   legacy: false
>   debounce: 5
>   qos: 2
>   match:
>       - vendor: IKEA
>         model: ['E1524/E1810',  'E2001/E2002', 'E1743']
>         options:
>           debounce: 0
> ```

how would we get different match/option sets? As the key/Val under device_options have to be unique.

I guess it could be ‘match_<free_to_choose_unique_label>’ and just check if the key stars with match_ ?

Koenkk, 2022-07-19:
`match` will be an array so the following will work:

```yaml
device_options:
  legacy: false
  debounce: 5
  qos: 2
  match:
      - vendor: IKEA
        model: ['E1524/E1810',  'E2001/E2002', 'E1743']
        options:
          debounce: 0
```

sjorge, 2022-07-19:
That would still not allow for matching day 3 models of hue remote and 3 models of ikea remotes and apply the same setting.

And yes, this is hard. Getting a **syntax** to do the matching specifically.

sjorge, 2022-08-19:
@Koenkk I still want to somehow implement this, but I don't really see a clean way to fold this into devices_options key given we want some sort of flexible matching. Did you come up with something?

Koenkk, 2022-08-19:
Not yet, I think it is also a good idea to wait a bit for input from other users. (such we consider more use-cases)

sjorge, 2022-08-19:
Perhaps we can tag it as Feature request so stalebot stays away from it?

Koenkk, 2022-08-19:
good idea, added the `dont-stale` label