Koenkk/zigbee2mqtt #13026: TuYa TS0601 temperature sensor humidity off by 10x

meeb, 2022-07-03:
### What happened?

Some recently purchased TuYa TS0601 sensors which identify as `_TZE200_bjawzodf` report the humidity incorrectly by a factor of 10x. For example the humidity is 56.1%, the display will show 56.1% however the MQTT payload contains 561%.

Log (this should be `"humidity":67.7` not `"humidity":677`):

```
Zigbee2MQTT:info  2022-07-03 21:09:30: MQTT publish: topic 'zigbee2mqtt/garage_temperature_sensor', payload '{"battery":100,"device":{"applicationVersion":67,"dateCode":"","friendlyName":"garage_temperature_sensor","hardwareVersion":1,"ieeeAddr":"0xa4c13855f58cb019","manufacturerID":4417,"manufacturerName":"_TZE200_bjawzodf","model":"TS0601_temperature_humidity_sensor","networkAddress":39047,"powerSource":"Battery","stackVersion":0,"type":"EndDevice","zclVersion":3},"humidity":677,"linkquality":47,"temperature":12.6}'
```

I'm using the zigbee2mqq Docker image, this occurs under both `latest` (1.26.0) and `latest-dev` (1.26.0-dev).

### What did you expect to happen?

The humidity should be returned as the actual reading, not 10x.

### How to reproduce it (minimal and precise)

Buy a TuYa TS0601 (_TZE200_bjawzodf) and connect it to a Sonoff Zibee 3.0 USB adapter with zigbee2mqtt.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB

Koenkk, 2022-07-04:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-04 by Koenkk

meeb, 2022-07-05:
Awesome, thanks!

kkossev, 2022-08-23:
@meeb have you noticed if this device is too chatty, sending the temperature, humidity, and battery level every 67 seconds?

meeb, 2022-08-24:
Yes, my `_TZE200_bjawzodf` devices do send a zigbee update message every minute or so each.

kkossev, 2022-08-24:
Thanks for the confirmation. This will definitely affect the battery life and will add more unnecessary load on the Zigbee mesh, but... it is what it is with these cheap devices.