Koenkk/zigbee2mqtt #13028: Most lights report being on after a restart of Z2M

dof250, 2022-07-03:
### What happened?

Each time I restart Z2M or HomeAssistant most of my lights are reported on. In reality they are off. It happens every time I restart Z2M or HomeAssistant. All lights report the same brightess of 69%. 

When restarting for a small period the lights have the correct state but then suddenly they all report being on, while still being off.

### What did you expect to happen?

I expect the lights to remain in the state they where before the restart

### How to reproduce it (minimal and precise)

Happens every restrart

### Zigbee2MQTT version

1.26

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0

dof250, 2022-07-05:
> same here

Wat version of Home Assistant are you running?

dof250, 2022-07-06:
Another thing I've noticed. When all lights are turned off and I restart, all lights stay off as expected. When just one light is on it seems like 80% off the lights take over the state of that one light.

deviantintegral, 2022-07-06:
I am seeing this as well. I can actually trigger it just by restarting Home Assistant too. I saw it on Home Assistant 2022.07.0 as well before I reverted due to an unrelated issue.

deviantintegral, 2022-07-06:
One more bit; if I hit the refresh button for an on / off switch in Z2M, it updates to the correct "off" state. I think the issue is entirely between z2m and HA, and not part of the zigbee layer.

Koenkk, 2022-07-07:
Can you provide a debug log of the following:
1. Start z2m
1. Turn off a few lights (note down which one)
1. Restart z2m
1. Now the light from step 2 should report as on

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

smarthomejunkie, 2022-07-09:
Same here. When I restart Zigbee2MQTT (in Home Assistant) all my Zigbee lights turn on. 
Turned off all my automations in Home Assistant to make sure that it was not in there, but that did not change anything.

smarthomejunkie, 2022-07-09:
> Can you provide a debug log of the following:
> 
> 1. Start z2m
> 2. Turn off a few lights (note down which one)
> 3. Restart z2m
> 4. Now the light from step 2 should report as on
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

I turned off my garden spots and then restarted Z2M.

I think you need these lines from the log:

Zigbee2MQTT:debug 2022-07-09 12:22:46: Received MQTT message on 'zigbee2mqtt/Garden Spot 01/set' with data '{"state":"ON","transition":1.0,"brightness":254,"color_temp":185}'
Zigbee2MQTT:debug 2022-07-09 12:22:46: Publishing 'set' 'brightness' to 'Garden Spot 01'
Zigbee2MQTT:debug 2022-07-09 12:22:46: Received MQTT message on 'zigbee2mqtt/Garden Spot 02/set' with data '{"state":"ON","transition":1.0,"brightness":254,"color_temp":185}'
Zigbee2MQTT:debug 2022-07-09 12:22:46: Publishing 'set' 'brightness' to 'Garden Spot 02'

Please let me know if you need more

smarthomejunkie, 2022-07-09:
It seems that Mosquitto broker sends an ON message every time when Zigbee2MQTT restarts. To me, it seems that Mosquitto stores a persistent state somewhere. (not sure, I'm lacking knowledge on this subject)

Koenkk, 2022-07-09:
@smarthomejunkie probably your MQTT broker has some retained messages. You can clear these by connecting to your MQTT server with any MQTT client and publish an empty **retained** message to `zigbee2mqtt/Garden Spot 01/set`

dof250, 2022-07-09:
@Koenkk I've pasted my debug log here:
https://pastebin.com/23nTWGMh
This is a few seconds after a restart. The first few seconds the status is correct. Light "kantoor" was at 25%. Then suddenly at 21:37:01 it published all lights off and after that all on. While still only "kantoor" was on at 25%

smarthomejunkie, 2022-07-10:
> zigbee2mqtt/Garden Spot 01/set

Well, that didn't work, unfortunately. I did this in Home assistant by calling the service mqtt.publish and filled in this in the topic field: zigbee2mqtt/Hall Downstairs/set (for example). I am missing some knowledge here, so it could be that I filled in the wrong field. Any thoughts?

darootler, 2022-07-11:
I don't know if that is related to the topic but i have an similar issue:

If i restart HA and just one light is on, all my lights turn on. If i restart HA and no light is on, all my lights stay off.

Regards
Richard

stickpin, 2022-07-12:
@Koenkk here is the debug log from my z2m:
[z2m_debug.log](https://github.com/Koenkk/zigbee2mqtt/files/9093872/z2m_debug.log)

Thanks!

dof250, 2022-07-12:
@Koenkk this indeed fixed the issue!
Just switched to the dev branch and tried multiple restarts of Home Assistant and z2m. Worked every time.

Thanks!

closed 2022-07-12 by dof250

dof250, 2022-07-12:
@smarthomejunkie you could also switch to the dev branch. The switchover is really easy. 

- Install the dev branch add-on
- stop the stable add-on
- copy your current config from the stable add-on
- paste in dev add-on and start

All devices will be there and work.

smarthomejunkie, 2022-07-12:
I can confirm that the issue is also fixed in my setup! 🎉

Koenkk, 2022-07-12:
Awesome, thanks!

RobinBeismann, 2022-07-28:
Thanks @Koenkk, this fixed it for me too. Any idea when this goes into the stable?

Koenkk, 2022-07-29:
@RobinBeismann next release is on 1 August

gionimi, 2024-03-23:
My environment has the same problem when I add the following automation

alias: Teste Retorno
description: ""
trigger:
  - platform: device
    type: changed_states
    device_id: 11f1543337a026fc8bb955635c3acedd
    entity_id: b5ff5ba37573564fd7c4ae1b900f163e
    domain: switch
condition: []
action:
  - if:
      - condition: device
        type: is_off
        device_id: df1b040ee52afb84596c11c728062a15
        entity_id: ae68a4d230c4afa04738eb52ce156a49
        domain: switch
    then:
      - type: turn_on
        device_id: df1b040ee52afb84596c11c728062a15
        entity_id: ae68a4d230c4afa04738eb52ce156a49
        domain: switch
    else:
      - type: turn_off
        device_id: df1b040ee52afb84596c11c728062a15
        entity_id: ae68a4d230c4afa04738eb52ce156a49
        domain: switch
mode: single

I'm usign Z2M Edge

Please, any idea for the solution?