Koenkk/zigbee2mqtt #13032: postponed data with debounce option

King-anderson, 2022-07-04:
### What happened?

Hi Friends,
I have enabled the "debounce" option for my Xiaomi door sensor. I see the published data after enabling this option is reverted. when I open the door, z2m publishes close status.
how can I handle this issue?

### Zigbee2MQTT version

1.25.2

### Adapter firmware version

20211115

### Adapter

CC2530

McGiverGim, 2022-07-04:
A debug log can help with this.

King-anderson, 2022-07-06:
actually, the device sends its battery status each 1 hr. but I want to be sent one time in a day. howere, as you said, the debounce package sends previous data.
is there any other solution to satisfy my requirement?

closed 2022-08-14 by github-actions[bot]