Koenkk/zigbee2mqtt #13033: Namron 4512737 - Touch termostat

oagers, 2022-07-04:
### What happened?

Hi,
The clock on the termostat is not showing the correct. Looks like it is using GMT but I am in GMT+2. Is there a way to set this? 
According to the user manual from Namron: 
The device Time Cluster is client, if the bound device supports local time of Time Cluster, the device will acquire local
time from the bound device spontaneously under 3 situations: acquire after this device is added to a network, acquire
when the device already added to a network and power is reset, acquire every 24 hours .

The documentation for the device in here do not mention anything about time. https://www.zigbee2mqtt.io/devices/4512737_4512738.html

My Home Assistant has the right timezone and clock.

Any tips on how to correct this?

### Zigbee2MQTT version

1.26

### Adapter firmware version

2022

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus

oagers, 2022-08-08:
@Koenkk got time to look at this issue? Where does the devices get time from? There is no setting in the Zigbee2mqtt module for this.

Koenkk, 2022-08-08:
How do you run Zigbee2MQTT?

oagers, 2022-08-08:
I run it on av RPi4 with Zonoff zigbee 3.0 usb stick. I run Hass.io.

Koenkk, 2022-08-08:
If you ssh into hassio, do a `docker exec -it Z2M_CONTAINER_NAME sh` and execute `date`, is the correct time shown?

oagers, 2022-08-09:
I ran the command "docker exec -it container-id date" but first ran the "docker container ls" to list the id of the container. The container has the correct time set.

oagers, 2022-08-09:
Now the time on the termostat have even driftet so now it is 2 minutes wrong.

Koenkk, 2022-08-09:
> Now the time on the termostat have even driftet so now it is 2 minutes wrong.

That is very strange, I would expect it to be shifted by exactly 1 or multiple hours (timezone), not minutes

@owangen since you've added support for this device, does the correct time show up?

oagers, 2022-08-09:
I have to correct myself. The termostat is still GMT + 2 min. Since I live in Norway the display shows 2 hours and 2 min wrong.

owangen, 2022-08-09:
I'll look into it. Just looked at my thermostat, and can't see the time at all on the display. I guess there is some setting to make it show @oagers ?

owangen, 2022-08-10:
I have checked my thermostat and can confirm that the time on my device is wrong for me also. Looked briefly on the code and I assume that I have not implemented time sync correctly. I guess the device is set with correct time at init and then drift over time because time sync is not implemented. Will dig more into it when I have time.

oagers, 2022-08-10:
@owangen I have done some test, and it looks like the clock is set when termostat is paired (zigbee). Even if the termostat looses power it will not set clock again. Looks like that there are options for time sync in the termostat according to Namron's user manual.

owangen, 2022-08-10:
@oagers Could you please try to Reconfigure option on the device from the z2m GUI? If my assumption is correct it will then set correct time.

oagers, 2022-08-10:
@owangen That corrects the clock, but still issue with wrong time zone. Mine is 2 hours off in Norway. It is correct time on the HA and the container that runs zigbee2mqtt. Can it be that it runs GMT time?

owangen, 2022-08-16:
Looking at the debug information I can't see any sign that the thermostat requests the time from the coordinator (see [debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/9354918/debug.txt) for debug log when thermostat joins the coordinator first time). 
@Koenkk I found this issue https://github.com/Koenkk/zigbee2mqtt/issues/1214, looks to me that that issue is not completed. Could this be of relevance when it comes to not time syncing with coordinator?

Koenkk, 2022-08-17:
Can you provide the herdsman debug log when pairing the device until the incorrect time is shown on it?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

owangen, 2022-08-17:
I thought the log from herdsman was present in the provided log file from my previous comment? 
In my original convert code I force a write to the time cluster at init, I think that is wrong thing to do so I tested now without those lines of code thinking that the thermostat requests the time from the coordinator.
The documentation for the thermostat states :
"The device Time Cluster is client, if the bound device supports local time of Time Cluster, the device will acquire local time from the bound device spontaneously under 3 situations: acquire after this device is added to a network, acquire when the device already added to a network and power is reset, acquire every 24 hours ."

I have not tried to sniff what the thermostat actually is requesting from the coordinator.

owangen, 2022-08-18:
Sorry for confusion. I realize now that since the documentation says that "the device Time Cluster is client", I guess this means that the coordinator, is Time server, and must send time to the thermostat. Is this correct @Koenkk. 
And if so, does Z2M have functionality to schedule sending updated time to device with given intervals, e.g every 24 hours?

Koenkk, 2022-08-18:
> I thought the log from herdsman was present in the provided log file from my previous comment?

I missed this indeed, but I don't see the read on `genTime` here. I expect that the device will read the time under the 3 situations as you mentioned. Z2M will then respond accordingly ([code](https://github.com/Koenkk/zigbee-herdsman/blob/6d43a67eefc5dd97641014c11a5295c3b6e518be/src/controller/model/device.ts#L219)).  

> acquire when the device already added to a network and power is reset

Can you trigger this situation and see if a read requests appears in the log (search for `genTime`)

owangen, 2022-08-18:
> I missed this indeed, but I don't see the read on `genTime` here. I expect that the device will read the time under the 3 situations as you mentioned. Z2M will then respond accordingly [...]

Thanks. The log contains everything happening from the thermostat joins to Z2M reports it's successfully configured. So I thought that the first situation ("acquire after this device is added to a network") should be fulfilled.

owangen, 2022-08-18:
I have searched some more and found this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5409 its for Sunricher zg9092 which Namron thermostat is white labeled from (although Namron devices have same modelid and souch) anyway it says:
"Time cluster doesn't seem to sync with coordinator, even after binding. We might need to write the time to it, as with some of the other thermostats. Manual mentions client cluster, but descriptor reports server cluster. I think maybe the binding needs to be on the time server, but I don't think the coordinator device/deCONZ supports this."
I guess Z2m also does not support binding device to time server? If so best fix will be to set correct time at configure, it's not a perfect solution because the clock on the device will drift over time

owangen, 2022-08-19:
I created a pull request to fix it at least on configure. The device does not support timzone or timestatus.
Ideally we should send correct time every 24 hours to prevent time drift on device

Koenkk, 2022-08-20:
Merged the fix

Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-20 by Koenkk