Koenkk/zigbee2mqtt #13034: Namron Luna 28W

oagers, 2022-07-04:
### Link

https://www.namron.com/products/namron-zigbee-luna-takarmatur-led-28w/

### Database entry

{"id":13,"type":"Router","ieeeAddr":"0x5c0272fffe578d55","nwkAddr":22909,"manufId":4456,"manufName":"LDS","powerSource":"Mains (single phase)","modelId":"3308431","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096,64642],"outClusterList":[10,25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":101,"hwVersion":1,"dateCode":"20191210-244","swBuildId":"2.15","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1656943208414,"defaultSendRequestWhen":"immediate"}

### Comments

Struggling with the js file, not sure about the definitions. The lamp (device) show up as LDS

### Supported color modes

Not sure, but looks like there are different leds.

thfthfthf, 2022-08-04:
Did you find a working solution for this?

Koenkk, 2022-08-04:
should work with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['3308431'],
    model: '3308431',
    vendor: 'Namron',
    description: 'Luna ceiling light',
    extend: extend.light_onoff_brightness_colortemp(),
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

If so, please provide the color temp range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

thfthfthf, 2022-08-04:
> Hello
>
> Thank you for your email!
>
> I am quite new to Home Assistant, so I am not sure where to put this
> configuration yet. I have a lot of reading to do.
> But if I understand correctly, I need to create the "ext_converter.js" in
> the "../config/zigbee2mqtt" folder?
> And then refer to in the configuration file for Zigbee2MQTT?
>
> The color temp range according to Namron is 2700 - 6500 K.

Hi again

I added this config according to your instructions and it worked!
One comment: Nothing happens after 370 mired in color temp, this is the
warmest setting.
[image: Screenshot 2022-08-04 at 19.01.56.png]

Thanks a lot for you support!

thfthfthf, 2022-08-04:
The color temp range for the light is according to Namron 2700 - 6500 K.

Koenkk, 2022-08-04:
@thfthfthf what is the coolest mired? https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device step 3.1 is the easiest way to find out.

thfthfthf, 2022-08-04:
@Koenkk Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}

Koenkk, 2022-08-04:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-08-04 by Koenkk

thfthfthf, 2022-08-04:
Awesome! Thank you.

oagers, 2022-08-08:
Where do I put the 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":370} in the  ext_converter.js file?

Koenkk, 2022-08-08:
@oagers yes that is correct.

thfthfthf, 2022-10-11:
Hello

Thank you for your email!

I am quite new to Home Assistant, so I am not sure where to put this
configuration yet. I have a lot of reading to do.
But if I understand correctly, I need to create the "ext_converter.js" in
the "../config/zigbee2mqtt" folder?
And then refer to in the configuration file for Zigbee2MQTT?

The color temp range according to Namron is 2700 - 6500 K.