Koenkk/zigbee2mqtt #13036: [New device support]: 9290031346 Hue Downlight (5996511U5 equivalent?)

ClintElliotMalcolm, 2022-07-04:
### Link

https://www.philips-hue.com/en-us/p/hue-white-and-color-ambiance-downlight-5-6-inch/046677578671

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0x001788010cd521f6","nwkAddr":37315,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCD006","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LCD006","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":2,"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20211210","swBuildId":"1.93.7"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20211210","swBuildId":"1.93.7","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1656947807200,"defaultSendRequestWhen":"immediate"}

### Comments

I believe this is the same as the Philips Hue 5996511U5 or at least another hue retrofit 5/6in recessed downlight. I bought 2 from a single box and those worked as expected, but a 4 pack came with these. The label on the inside of the light is different with the label.

FCC ID: 2AGBW9290031346X 
IC ID: 20812-31346x
Probably model #: 9290031346

### External converter

```shell
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
// Make sure extend.light_* is not used (hueExtend should be used instead)
const extendDontUse = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');

const hueExtend = {
    light_onoff_brightness: (options={}) => ({
        ...extendDontUse.light_onoff_brightness(options),
        ota: ota.zigbeeOTA,
        meta: {turnsOffAtBrightness1: true},
        toZigbee: extendDontUse.light_onoff_brightness(options).toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp: (options={}) => ({
        ...extendDontUse.light_onoff_brightness_colortemp(options),
        ota: ota.zigbeeOTA,
        meta: {turnsOffAtBrightness1: true},
        toZigbee: extendDontUse.light_onoff_brightness_colortemp(options).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_color: (options={}) => ({
        ...extendDontUse.light_onoff_brightness_color({supportsHS: true, ...options}),
        ota: ota.zigbeeOTA,
        meta: {turnsOffAtBrightness1: true},
        toZigbee: extendDontUse.light_onoff_brightness_color({supportsHS: true, ...options}).toZigbee
            .concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
    light_onoff_brightness_colortemp_color: (options={}) => ({
        ...extendDontUse.light_onoff_brightness_colortemp_color({supportsHS: true, ...options}),
        ota: ota.zigbeeOTA,
        meta: {turnsOffAtBrightness1: true},
        toZigbee: extendDontUse.light_onoff_brightness_colortemp_color({supportsHS: true, ...options})
            .toZigbee.concat([tz.hue_power_on_behavior, tz.hue_power_on_error]),
    }),
};

const definition = {
    zigbeeModel: ['LCD006'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '9290031346', // Vendor model number, look on the device for a model number
    vendor: 'Philips', // Vendor of the device (only used for documentation and startup logging)
    description: 'Hue white and color ambiance 5/6" retrofit recessed downlight', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: hueExtend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

### Supported color modes

color temperature, rgb

ClintElliotMalcolm, 2022-07-05:
Well that's frustrating. Got it to work, but it only responded if I had it read one at a time (I believe).

Anyways: "colorTempPhysicalMin":153,"colorTempPhysicalMax":500

Koenkk, 2022-07-06:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-07-06 by Koenkk