Koenkk/zigbee2mqtt #13037: Legrand "Dimmer switch with neutral" L441C/N4411C/NT4411C refresh status

Loic691, 2022-07-04:
### What happened?

2 problems :
The first as minor :
- Legrand 067797 simple switch (or gradable) switch is seen as "L441C/N4411C/NT4411C".

The second as major :
- When we change the state of switch with physical button, the state is not update. We have to click on refresh button for this.

### What did you expect to happen?

I expected that the refresh state have to be refresh in "real time"

### How to reproduce it (minimal and precise)

When switch is off, click on the physical button to switch on the switch.
The physical lamp is going UP.
But the state in the J2M web GUI is not refresh

### Zigbee2MQTT version

1.2.6.0

### Adapter firmware version

0x26720700

### Adapter

conbee2

closed 2022-08-12 by github-actions[bot]

Loic691, 2022-10-23:
Yes it s working.
I had to pair again the device.
Sincr this post i changes my usb coordinator to sonoff ZB Dongle E.

kevincaradant, 2022-10-23:
alright good news :)
In fact, this reference is not mentioned in the documentation of Z2M as Legrand ? 

It's recognized into https://www.zigbee2mqtt.io/devices/L4411C_N4411C_NT4411C.html#bticino-l4411c-n4411c-nt4411c insted of https://www.zigbee2mqtt.io/devices/067773.html ?

and you use it with neutral wire ? 
In dimmer mode ? 

Thanks

kevincaradant, 2022-10-23:
```
Power_on_behavior (enum)
Controls the behavior when the device is powered on. Value can be found in the published state on the power_on_behavior property. To read (/get) the value publish a message to topic zigbee2mqtt/FRIENDLY_NAME/get with payload {"power_on_behavior": ""}. To write (/set) a value publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"power_on_behavior": NEW_VALUE}. The possible values are: off, previous, on.

[#](https://www.zigbee2mqtt.io/devices/067771.html#linkquality-numeric)
```

This feature is not mentioned in your L4411C.

Do you know what is the behavior when the power is down and then up ? 

Thanks

Loic691, 2022-10-24:
> It's recognized into https://www.zigbee2mqtt.io/devices/L4411C_N4411C_NT4411C.html#bticino-l4411c-n4411c-nt4411c insted of https://www.zigbee2mqtt.io/devices/067771.html if I read your thread ?

yes. I don't understand why...

Loic691, 2022-10-24:
> and you use it with neutral wire ? In dimmer mode ?

Yes

kevincaradant, 2022-10-24:
OK thanks. I will report that because for me it's an issue. 

and about my latest question. What's the behavior on power outage? I don't want something which power on auto during a poweroutage at night 

Thanks

Loic691, 2022-10-26:
As explained. I don't know for the moment. I will have to check and confirm you

kevincaradant, 2022-12-11:
@Loic691 
Hi

Any news about the power outage ? :)
I would know if we can choose the behavior or at least it go OFF by default :)

Thanks

Loic691, 2022-12-22:
In z2m front end there is no way to change the power behaviour.
Is the module can with legrand specific information in the legrand gateway ? I don't see that after pairing with legrand when i updated the module

So @kirovilya I have always the problem as others users of the first part of this issue.
So the module isn't recognize with the good  name and reference.

I have no idea to where where change this information... I think whe can't...

Loic691, 2022-12-22:
Sorry it was not clear.
I updated my post

kevincaradant, 2022-12-22:
> Sorry it was not clear. I updated my post

Thanks , the behavior can't be changed ( for the moment ) because as you mentionned, it's not the right device recognized.
But what's the behavior currently existing on power outage if you can simulate it ? 

ON / OFF / restore the latest state known ? :) 

Thanks ;)

Loic691, 2022-12-22:
I just checked. Sorry for delay
The behaviour is RESTORE THE LAST STATE
In my mind, this is a good behaviour.

kevincaradant, 2022-12-22:
Ok perfect it's what I expected as behavior. Thanks for your confirmation.
Last issue, just got the right device as detection during pairing now ^^