Koenkk/zigbee2mqtt #13038: GLEDOPTO GL-C-006P is seen as GL-D-005P and the cold/warm white is not working

Loic691, 2022-07-04:
### What happened?

The GL-C-006P is seen as DL-D-005P which is a RGB LED instead of a cold/warm light.
So the settings of color temperature of the controleur not working property

### What did you expect to happen?

I expected that the settings of warm and cold white must be correctly manage.
No RGB settings would be appears on this device

### How to reproduce it (minimal and precise)

Just include the controler to see the modelID is not correct.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

0x26720700

### Adapter

Conbee2

Loic691, 2022-07-04:
So a strange update.
I tryed another GL-C-006P wich is seen correctly and worky perfected.
So the only problem is my old device GL-C-006P is not seen correctly.
I tryed to reset it and reinclude but it the same. I think ther is a firmware problem on the device...
I updated the issue

Koenkk, 2022-07-05:
This is an issue with the firmware of the device, stopping z2m and manually changing the modelID in the data/database.db is a workaround.

closed 2022-08-14 by github-actions[bot]