Koenkk/zigbee2mqtt #13039: Zigbee Network Unstable - Goes Offline Sporadically until z2m container restart

Cossey, 2022-07-04:
### What happened?

In the past 1-2 months, my Zigbee network has sporadically gone offline, until I restart z2m. In some cases, I also have to restart some of my routers (assortment of Philips Hue lights). It will run okay but then I will start getting `SRSP - AF - dataRequest after 6000ms` with one of my lights, in the attached log case it starts with `Hallway2Light` going offline, then all the other devices start going offline. 

My battery powered devices still show as Online during this phase, however they are technically offline as they don't send data (for example, my Phillips Hue Motion sensors, display the long red light, indicating problems communicating).

In this most recent occurrence, I restarted on the 27/06 (when I last had this issue) and the network started to fail on 05/07 around 3:42am. So this time it ran for a week without issue - however I have had it fail previous within a few hours and in another case overnight, so the timing does not seem consistent. Restarting the z2m docker container brings everything back online.

Originally I had my dongle connected it up to a 1m cable, but now I have my dongle connected to a 3m USB cable - BOTH setups it has failed in the same way, so I have ruled out the cable length.

Although I cannot be certain, I felt that this issue started to appear once I updated `zStack3x0` to `20220219` and `z2m` to `1.25.2`. I have used previous z2m that I thought did not have this issue, but it still occurs. Now I have updated to latest zstack beta `20220507` in case it was a problem with 0219. 

_Zigbee channel and WiFi do not overlap._

| Key | Value |
| - | - |
| Zigbee Devices | 50 |
| Zigbee Channel | 11 |
| USB Autosuspend | -1 (disabled) |
| Docker | 20.10.14 (linux/amd64) |
| Debian | 11 bullseye |
| System | HP Prodesk Mini |

Any suggestions on how to diagnose or next steps would be greatly appreciated.

### What did you expect to happen?

The network should stay stable and keep going "till the end of time."
I had no issues when originally running on the Raspberry Pi 3, and for quite some time this HP Mini.

### How to reproduce it (minimal and precise)

I can say exactly how to reproduce the issue, as it happens sporadically without any obvious indicators.

### Zigbee2MQTT version

1.25.2-dev commit: 8adb2f1

### Adapter firmware version

20220507

### Adapter

Sonoff ZB Plus Dongle

### Debug log

Debug Log is approximately 256MB, I cannot attach directly here due to size limitation so uploaded to dropbox:
https://www.dropbox.com/s/a942vurauiwz6sg/zigbee.log?dl=1

Failure appears to begin at `Line 1503956` at `2022-07-05 03:42:57` with `Failed to ping 'Hallway2Light'`

_Please note that `BackroomLight` goes offline often due to mother-in-law often turning this light off via the light switch instead of hue dimmer, so this is expected._

Cossey, 2022-07-05:
It has just happened again, this time it only lasted 5 hours before everything went offline again.

[zigbee2.log](https://github.com/Koenkk/zigbee2mqtt/files/9042996/zigbee2.log)

csowada, 2022-07-05:
Same issue here with following system in a docker container on Esxi. It worked very well in the past, but since several weeks it stops after weeks or just several hours. One major change in the past was the "Available" feature that I've enabled.

Zigbee2MQTT Version: 1.26.0-dev commit: [53ae00c](https://github.com/Koenkk/zigbee2mqtt/commit/53ae00c)
Coordinator Typ: zStack3x0 (sonoff Zigee 3 Pro Stick)
Coordinator Version: 20211217

Koenkk, 2022-07-05:
> Although I cannot be certain, I felt that this issue started to appear once I updated zStack3x0 to 20220219 and z2m to 1.25.2

Does downgrading the firmware fix the issue?

Cossey, 2022-07-06:
Okay, I have downgraded to `20210708`, if I recall I think that came with the adapter when I first got it and its a version lower than what @csowada reported above. 

I would assume that if the issue does not occur in the 4 weeks then everything is fine. Is there a way to make sure this issue doesn't go stale until then? I will of course report sooner if we have issues, with the logs.

Cossey, 2022-07-06:
Okay even with downgraded to `20210708` it still seems to have problems.
I have also did a `dist-upgrade` on my Debian host to see if anything changed, but it was still the same issue after reboot.

Right now, I am running a Zigbee packet capture alongside the logging, hopefully this will help in diagnosing the problem. After I get a capture of this, hopefully it would shed some light. I started capturing earlier today, but the device doing the captures went into standby by mistake - on resume however, there were tons of packets reporting:
```
Command Frame: Network Status
    Command Identifier: Network Status (0x03)
    Status Code: No Route Available (0x00)
    Destination: 0x0000
```

So I will aim to get a full capture from start to crash and beyond. 

After, I will try going back a few versions of z2m to see if it could be that - although I can't go back too far due to support for one of the devices needing specific minimum version.

Cossey, 2022-07-06:
Yes, I will try disabling that and seeing how it goes. 

I did however get a full capture from the Zigbee network when it failed again overnight [220706 Capture.zip](https://github.com/Koenkk/zigbee2mqtt/files/9059149/220706.Capture.zip), if that might help.

The password will be sent via Telegram (Encrypted using 7zip AES-256).

Cossey, 2022-07-07:
Disabling Availability did not help - it still crashed last night.

Cossey, 2022-07-07:
Seems like its consistently failing every 2-5 hours now. Had restarted container (including dongle being physically removed) at least 5 times today. Its practically unusable now.

Koenkk, 2022-07-07:
@Cossey I would suggest to first switch back to the Pi, if it keeps on working there we can exclude z2m or the stick being the problem.

Cossey, 2022-07-08:
Okay I did go back to the Pi, and the same thing happened. Also tried a smaller extension cable just incase on the Raspberry Pi.

I think I might just clear out all the data in the container and repair from scratch (Ill just keep the mqtt names of the devices and the settings file).

Cossey, 2022-07-14:
Okay, so I wiped out all of the files used in the docker container except for the `configuration.yml` and just changed the pan id by 1 and then modified the network key slightly and re-paired devices (bar one Aqara THP that won't pair until a get a new CR2032) and so far it has been stable for 5 days. This was on the HP system, not the Rpi. 

I think it may either be this Aqara THP that I can't yet re-pair, or something with the state of the `state.json`, `database.db`, `coordinator_backup.json` files that was causing an issue.

I kept a backup of those 3 files before I removed them from the container disk binding.

I don't like to make a habit of re-pairing all 50 devices though fortunately all are easily accessible, and most paired without issue (though I did have issue with 2 of the older style hue dimmers and two of the Aqara THPs).

I am hoping that this issue won't reoccur again anytime soon. Before it became a daily problem it was occurring every 2-5 weeks. Hopefully it will stay stable until the hardware dies.

Koenkk, 2022-07-14:
@Cossey good to hear it is stable now. Did you ever migrate from the CC2531 to the sonoff dongle? If yes, was the `coordinator_backup.json` removed before this?

Cossey, 2022-07-14:
No, the Sonoff is the first and only Dongle I've used with z2m.

I have updated the Sonoff firmware 4 or so times, and I have to use the TI Flash Programmer 2 with pressing the onboard button throughout since the Bootloader doesn't seem to work anymore.

Our location has also had a couple of powercuts over the course of using z2m.

Is it advisable to update the firmware and z2m container at the same time or should this be done in two phases?

Koenkk, 2022-07-15:
> Is it advisable to update the firmware and z2m container at the same time or should this be done in two phases?

No, should be fine to update both at the same time.

thesilentmiaow, 2022-08-06:
I've been getting the same / similar. Randomly seeing all the routers go offline and each one logging:
{"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'

Restarting the addon resolves it or alternatively just leaving it for several hours and it eventually sorts itself out before happening again usually after a day or so. Running HA OS on RPI4.  I was on zStack3x0 v20211217 and have just upgraded to v20220219 and swapped out the coordinator in case it was a hardware issue but the problem started again within an hour.

Koenkk, 2022-08-07:
@thesilentmiaow sounds like that the stick is disconnected somehow, check your Pi power supply and make sure the stick is not connected via a hub

thesilentmiaow, 2022-08-08:
> @thesilentmiaow sounds like that the stick is disconnected somehow, check your Pi power supply and make sure the stick is not connected via a hub

Currently has the official Raspeberry Pi 3A power supply plugged in directly (no hub) although I'm running an SSD and the Sonoff Zigbee stick from USB so could be hitting a limit.  I've ordered a 3.5A supply and will see if that makes a difference.

Koenkk, 2022-08-08:
>  I'm running an SSD and the Sonoff Zigbee stick from USB 

This is indeed a common problem: https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#raspberry-pi-users-use-a-good-power-supply

Cossey, 2022-08-09:
I'd like to report that so far, my Zigbee network has been stable, based on what I did in this [comment](https://github.com/Koenkk/zigbee2mqtt/issues/13039#issuecomment-1184044589). I should also add that I did finally pair my Mi Temperature, Humidity and Pressure sensor a few weeks ago without a seemingly negative impact.

I can only conclude that the state of the container itself got out of sync with network/dongle and began to cause issues which eventually got so bad that it would take down the Zigbee network every 1-5 hours. 

Ill close this issue for now, but if it happens again, this would point to something more going on behind the scenes.

Cossey, 2022-08-24:
So bad news, my Zigbee network went down last night out of the blue.

When this happened, I replaced some batteries on a few of my devices and restarted the z2m container... and this morning I woke up and the network was offline. There is something more going on here. Obviously I want to avoid having to re-pair 50 devices every 41 days to get it to work again.

@Koenkk what can you suggest? Unfortunately debug mode was no longer enabled but I have restarted the container with debug mode on.

I can state with certainty that there was no power cuts throughout this time, although I did take down the container last week (its a stack with openhab and mosquitto so I do take it down occasionally to update all the images).

Koenkk, 2022-08-25:
@Cossey without debug logging I cannot determine what the root cause was. If it happens again please provide me the herdsman debug logging.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Cossey, 2022-08-25:
[zigbeecrashaugust.log](https://github.com/Koenkk/zigbee2mqtt/files/9428412/zigbeecrashaugust.log)

This is from two separate runs over the past few days where it failed.

umrath, 2022-08-26:
Looks like I have a similar issue.
During this night huge parts of my devices (about 1/3 of > 100) suddenly went offline.

Every try to fix this failed until I restarted z2m. All devices went back to online within 2 minutes.

I had single devices marked as offline occasionally (maybe one device per week) and could fix this by clicking reconfigure.
This is the first time that many devices showed the same behaviour at the same time.

Koenkk, 2022-08-26:
@Cossey it looks like the dongle just disconnects. Since you are running your PI from an USB SSD, did you connect the SSD via an external powered USB hub?

Cossey, 2022-08-26:
I'm not running this on my pi or with a SSD? I am running this on an x64 HP Mini PC with 8GB of RAM. The PC has a m.2 drive installed.

Koenkk, 2022-08-27:
I see, can you provide me a sniff of your network. Preferably one with the fail in it, but if that is not possible a few hour long sniff will also do it. I want to see if I can find some strange traffic on your network.

Cossey, 2022-08-27:
Thanks Koen, I don't have a sniff of these recent events yet, however when I originally had this issue over a month ago i had logs and traffic sniffs, please refer to those - they are further up in the comments. I will endeavour to get a sniff when it occurs next. 

This is the same issue with the exact same behaviour. I will also resend the compressed file password over telegram. 

I'm sure its not the specific dongle, as I swapped out with another ZDP which I haven't changed the firmware from factory. 

All routers are Phillips Hue except the Garage Door Opener which is a Tuya.

My Access Point is on channel 6, Zigbee network on channel 11.

Repairing all devices around 44 days ago solved the problem till just the other day. I do have 50 devices on the Zigbee network but with all the routers it should be able to handle this.

Cossey, 2022-08-28:
@Koenkk here are logs, configuration and Zigbee Traffic Sniff from today. Failed at around 7:00-8:00pm NZ Time. Network key in the config yaml (the key is different to the one I sent to you today, due to repairing devices last month and changing key/pan ID).
[Aug28.zip](https://github.com/Koenkk/zigbee2mqtt/files/9439322/Aug28.zip) Password is the same.

Thanks so much for looking at this Koen. All I can say is that restarting the docker container brings it back online for a 5 minutes to 14 hours, with no consistent timing.

Some things to note:
- Swapping the Sonoff dongle with another Sonoff of the same type still had the error occurred. 
- Shifting the container, configuration and device to the Raspberry Pi 3 from the x64 HP PC didn't change anything, the error still occurred.
- When this network crashes/fails Mains Powered Devices show as offline in the Web UI - however battery devices still show online - but they failed (Philips Hue Motion sensors show long-red LED, Philips Dimmers show Red LED, Xiaomi Button events/Motions sensors aren't received, Sonoff Motion sensors ).
- Clearing the NVRAM of the dongle on my desktop PC using the ZigpyZNP tools didn't make a difference when plugging back into the HP PC. Still crashed.

Koenkk, 2022-08-28:
I found something interesting. The last send message of the coordinator is highlighted in blue below. Shortly after that a strange `Coordinator Realignment` message is send, maybe this is what causes the problems. What kind of device has ieeAddr `0x00124b0025092368`?

[image]

Cossey, 2022-08-28:
Finally, something to go on. Thanks Koen! Its one of the newer devices I have connected to my network! It is a [SONOFF SNZB-03](https://www.zigbee2mqtt.io/devices/SNZB-03.html#sonoff-snzb-03). I have two of them.

[image]

This is what I will do - I will take this device offline and restart the container and see how long things fare until the device goes offline. If the network goes down I will remove both SONOFF SNZB-03 devices. I will consider the test successful if it manages to stay up for over 3 days straight!

Looking online, it seems like these devices are a hit and miss when it comes to problems people have been experiencing, more so depending on temperature. I know these are inside sensors however they both are outside on the roof soffit (underside of the roof overhang), but there is grass underneath and it has been cold here at night over the past few days. Funnily enough, I have no issues with the Aqara Motion Sensors outside (aside from one sensor which gives occasional false alarms at night - probably due to temp/position). 

If you have a chance, can you explain your process for going through the Zigbee Sniffs and determining issues - maybe even create a section on it on the z2m website when you have the time? I would like to know what to look for when these things happen.

thesilentmiaow, 2022-08-29:
So as per the advice given earlier I have upgraded my Pi's power supply to 3.5A and also now have the SSD connected via a powered USB hub.  Yet the problem is still occurring.  Same as @Cossey I also have a SNZB-03 which is outside in a sealed weather proof container by the backdoor (hole cut for the PIR motion sensor and sealed around the edge).  I also have a couple of SNZB-04 in the garden, again in sealed containers.  I think my next step will be to completely uninstall Z2MQTT and start from scratch but not add the SNZBs into the network just yet and see if the problem comes back.

Cossey, 2022-08-30:
I pulled the batteries out of both SNZB-03's without doing any changes (aside from container restart) and the network hasn't crashed for 30 hours so far (it should've crashed overnight at least). I would assume that this device is causing the issues... but the reason is why? I would have hoped the Coordinator would be able to withstand this. Could this be a bug in the TI stack?

Koenkk, 2022-08-30:
@Cossey that is great news, lets monitor for two more days. 

> Could this be a bug in the TI stack?

Yes definitely, I did a quick check on the source code and the handling of this command is in their closed source libs. If it still works for two more days I will create a post at the TI forum. Let me know.

Cossey, 2022-09-01:
So the network has been running stable for 3 days now.

Although my home is covered with Philips Hue Routers, the two motion sensors were near the Zigbee Coordinator, which is stuck to the wall of the south side of the house. So I assume the signal just wasn't quite good enough so the problem device needed to shift router.

I notice in the captures, the Link Status packets are routers broadcasting their visible neighbours, but I do not understand what the Incoming/Outgoing Cost is or how it is calculated. What the purpose of broadcasting this information, do other routers determine how to route using this information?

Can you confirm that I understand the capture correctly: 
Motion sensor was having issues connecting to its parent (lets assume the coordinator).
[82625] Broadcasts an orphan notification (which is Zigbee HA?). I assume that Broadcast means that any device that can hear it will route message to coordinator.
[82626] Coordinator sends realignment message across network, saying that device should parent up to coordinator 0x7424 (which is my Philips Hue Colour Light 9290012573A, technically a router). 
[82627] Acknowledgement sent (not sure if it comes from Router or End Device)
[82628] Device then announces itself across Zigbee Network (Noticed the sequence number is 0).
Then there is no acknowledgement of announce (sequence no 0).
[82630] Device then complains to router that announce timed out with no acknowledgement.
[82631] Hue Router acknowledges the timeout of the last acknowledgement.
Starting from [82632] is the SNZB trying to reconnect to the network by enrolling, the first few packets get acknowledged, but then at [82645] onwards it doesn't get acknowledged, hence the multiple packets with the same sequence number. I assume that this is where the Coordinator has crashed.

Koenkk, 2022-09-05:
> What the purpose of broadcasting this information, do other routers determine how to route using this information?

You can read more about it here: https://www.digi.com/resources/documentation/Digidocs/90002002/Concepts/c_zb_link_status_trans.htm?TocPath=Transmission%2C%20addressing%2C%20and%20routing%7CRF%20packet%20routing%7C_____1

> Can you confirm that I understand the capture correctly:

Yes that is correct

I've created a bug report at the TI forum, lets see if they can fix it: https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/1131804/simplelink-cc13x2-26x2-sdk-coordinator-crash-when-orphan-notification-coordinator-realignment-is-received

Cossey, 2022-09-05:
Thanks for following up and looking into this @Koenkk, much appreciated. 

@thesilentmiaow did removing the batteries for the Sonoff Motion Sensors fix you issue? I have in the past re-paired all devices including the Sonoff Motion Sensors... and it worked... for 41 days but then the issue appeared again. Either way please let us know of your progress.

closed 2022-10-14 by github-actions[bot]

rubdos, 2022-11-02:
I'm also experiencing this issue. Is there a way to restart or crash z2m when the USB dongle fails in this way? That way, `systemd` or `docker` can at least auto-restart the daemon. It's not like the z2m daemon is doing something useful when the USB dongle fails completely, so crashing the daemon actually makes sense, I would think.

rubdos, 2023-02-10:
FWIW, I have upgraded to CC2652RB_coordinator_20221226, and I still experience this issue. Is there any way we can help TI in fixing this, @Koenkk? This seems like a DoS vulnerability to me.

Koenkk, 2023-02-12:
@rubdos could you provide me the herdsman debug log of this? There are 2 reasons this can happen:
- the stick itself crashes
- the stick disconnects due to e.g. bad USB cables, another process trying to access the stick

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

rubdos, 2023-02-13:
I could be a few hours or days, but I'm catching the logs now. I'll drop you a link when I have them, because I expect it to be rather large :'-)

rubdos, 2023-02-13:
@Koenkk https://nextcloud.rubdos.be/index.php/s/jfTTmFPZdwtgdgY

It's a bit big, I hope I have gz-inlining turned on on my nginx :-)

Let me know when you have the file, I'd rather take it offline asap.

Koenkk, 2023-02-14:
Looks like scenario 2: the stick disconnects due to e.g. bad USB cables, another process trying to access the stick

In case of scenario 1 there are usually some `<-- AF - dataRequest - {"status":0}` where status is not 0 before the crash. 

The reasons of this vary a lot, bad power supply, other usb devices interfering, processes trying to access the port. I even saw one case where this was caused due to the power supply being connected to a smart wifi relay.

rubdos, 2023-02-14:
My stick (slae.sh) is directly plugged into the motherboard, and that's an X399M Taichi board of €500...  I truly doubt this. It only started happening after I integrated some Philips Hue spots into the network, before it ran perfectly stable. I can still try swapping it to another port or putting the cable back inbetween...